corna/me_cleaner #233: Clean an older core 2 quad

orachas, 2018-10-07:
I would like to run me_cleaner on an older HP (6000 pro sff) desktop running a Core 2 quad q9650 with a Q43 express chipset.

Running flashrom to extract the current bios fails (due to failed permission to read the ME?), but it tells me that I have an ICH10D with a Winbond flash chip "W25Q32.V." I only see one physical Winbond chip on this board, and it's not socketed (I don't see any socketed chips other than the CPU). Running mei-amt-check reports what I assume to be ME version 5.2.10.

I have the last HP BIOS from 2015 in a file called 7G2_0203.bin. I can generate an ifdtool dump which I'll attach, but me_cleaner reports "Unknown image" on it.

Are there any conversion tools that I can use on the OEM BIOS to create something that me_cleaner can patch? And can I convert this output back into something that the HP tools can write?

persmule, 2018-10-07:
Check https://github.com/corna/me_cleaner/issues/227 for reference.

orachas, 2018-10-07:
Thanks, I will try Libreboot's ich9deblob tools for this board instead.

corna, 2018-10-07:
This morning I've added support for ICH9 and older, no need to use ich9deblob. Try to pull the latest changes in me_cleaner and run it again on your image.

orachas, 2018-10-07:
Ok, here is the log of what I did:

    $ wget https://raw.githubusercontent.com/corna/me_cleaner/master/me_cleaner.py
    --2018-10-07 12:22:59--  https://raw.githubusercontent.com/corna/me_cleaner/master/me_cleaner.py
    Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.184.133
    Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.184.133|:443... connected.
    HTTP request sent, awaiting response... 200 OK
    Length: 37482 (37K) [text/plain]
    Saving to: ‘me_cleaner.py’

    me_cleaner.py       100%[===================>]  36.60K  --.-KB/s    in 0.02s

    2018-10-07 12:22:59 (1.68 MB/s) - ‘me_cleaner.py’ saved [37482/37482]

    $ chmod 755 me_cleaner.py

    $ ./me_cleaner.py -c 7G2_0203.bin
    Full image detected
    Found FPT header at 0xb010
    Found 14 partition(s)
    Found FTPR header: FTPR partition spans from 0xd2000 to 0x162000
    ME/TXE firmware version 5.2.10.1023 (generation 1)
    Public key match: Intel ME, firmware versions 5.x.x.x
    The meDisable bit in ICHSTRP0 is NOT SET
    The meDisable bit in MCHSTRP0 is NOT SET

    $ ./me_cleaner.py -S -O nome_6000pro.bin 7G2_0203.bin
    Full image detected
    Found FPT header at 0xb010
    Found 14 partition(s)
    Found FTPR header: FTPR partition spans from 0xd2000 to 0x162000
    ME/TXE firmware version 5.2.10.1023 (generation 1)
    Public key match: Intel ME, firmware versions 5.x.x.x
    The meDisable bit in ICHSTRP0 is NOT SET, setting it now...
    The meDisable bit in MCHSTRP0 is NOT SET, setting it now...
    Disabling the ME region...
    Wiping the ME region...
    Done! Good luck!

I have three options to burn this...

- The Windows 7 flash tool, which I am running in Windows 10.
- A FlashBin.exe tool in a "DOS Flash" directory included with the above Win7 set.
- A flash update option in the BIOS that supposedly reads from USB.

Which should I try first?

orachas, 2018-10-07:
I have flashed this image successfully within the HP BIOS, powered down, and cleanly rebooted.

How can I verify that the ME is disabled? I still see the output below:

    # ./mei-amt-check
    AMT present: true
    AMT provisioning state: being provisioned
    Flash:      5.2.10
    Netstack:   5.2.10
    AMTApps:    5.2.10
    AMT:        5.2.10
    Sku:        18462
    VendorID:   8086
    Build Number:       1023
    Recovery Version:   5.2.10
    Recovery Build Num: 1023
    Legacy Mode:        False

Assuming that I've successfully wiped my flash, I might write this procedure up.

orachas, 2018-10-09:
I have applied this to a few similar machines with no difficulty, and it seems to work well.

After removing all power from these systems for some time, the mei-amt-check utility is no longer able to find an active ME. It does seem that the ARC CPU is able to survive a short loss of power, maintaining the running ME instance across a reboot. I am guessing that ten minutes with no power is sufficient to reset it.

Am I correct in guessing that this is the last Intel architecture that is not vulnerable to SA-00086?

closed 2018-10-12 by orachas

Jolex12, 2018-12-01:
Can you please send me 7G2_0203.bin ?

orachas, 2018-12-01:
> Can you please send me 7G2_0203.bin ?

Below is the processed BIOS file that you requested. I'm assuming that you know how to undo the base64 encoding and xz compression. Please do check the signature.

Hewlett-Packard HP Compaq 6000 Pro SFF PC/3048h, BIOS 786G2 v02.03 10/19/2015
dbcbd10f057a39c617937a1a80c74e08dc83cde5d80952ae482c740d188bd5a4  7G2_0204.bin
[7G2_0204.bin.xz.base64.txt](https://github.com/corna/me_cleaner/files/2635988/7G2_0204.bin.xz.base64.txt)

I also have a processed BIOS for the HP dc7800p SFF, attached below.

26d4854102d6f2383aa33e72ff64905d62ec23e9322708f3b75b11e7901f8b21  7F1_0136.bin
[7F1_0136.bin.xz.base64.txt](https://github.com/corna/me_cleaner/files/2635992/7F1_0136.bin.xz.base64.txt)

Hope they work as well for you as they did for me.

Jolex12, 2018-12-05:
I'm really sorry to bored you,but can you please send me
7G2_0204.bin.xz.base64.txt in bin file.
I don't now how change it to bin file.
Greeting from Bosnia ! :)

orachas, 2018-12-07:
> I'm really sorry to bored you,but can you please send me 7G2_0204.bin.xz.base64.txt in bin file. I don't now how change it to bin file. Greeting from Bosnia ! :) суб, 1. дец 2018. у 16:33 orachas <notifications@github.com> је написао/ла:
> […](#)

If you don't know how to work with common archive formats, then are you sure that you want to run a custom BIOS?

In any case, the above .txt file can be decoded and uncompressed on Windows with the latest copy of busybox for win32 (which you should easily be able to find.

This command will generate the .bin file:

   busybox64 base64 -d 7G2_0204.bin.xz.base64.txt | busybox64 xz -cd > 7G2_0204.bin

This command will print the sha256 signature:

   busybox64 sha256sum 7G2_0204.bin

It must match this perfectly:

   dbcbd10f057a39c617937a1a80c74e08dc83cde5d80952ae482c740d188bd5a4  7G2_0204.bin