corna/me_cleaner #234: Lenovo Thinkpad T440p

precurse, 2018-10-14:
Hello,

I successfully pulled the BIOS image using the clip clamp method, using flashrom + Raspberry Pi. I verified the file hash 3 times and know the image is good.

However, when I try to run ME Cleaner on it, I get "unknown image":
```
$ python me_cleaner/me_cleaner.py -c bios.bin
Unknown image
```

I verified the image looks ok (to me). iftool reports:
```
Found Region Section
FLREG0:    0x5df97f78
  Flash Region 0 (Flash Descriptor): 00f78000 - 00df9fff (unused)
FLREG1:    0x3a69271e
  Flash Region 1 (BIOS): 0071e000 - 00a69fff 
FLREG2:    0xed43facb
  Flash Region 2 (Intel ME): 00acb000 - 00d43fff 
FLREG3:    0x8956ec05
  Flash Region 3 (GbE): 00c05000 - 00956fff (unused)
FLREG4:    0x6fe6a465
  Flash Region 4 (Platform Data): 00465000 - 00fe6fff 

FLMSTR2:   0x4c81839b (Intel ME)
  Platform Data Region Write Access: disabled
  GbE Region Write Access:           enabled
  Intel ME Region Write Access:      enabled
  Host CPU/BIOS Region Write Access: disabled
  Flash Descriptor Write Access:     disabled
  Platform Data Region Read Access:  disabled
  GbE Region Read Access:            disabled
  Intel ME Region Read Access:       disabled
  Host CPU/BIOS Region Read Access:  disabled
  Flash Descriptor Read Access:      enabled
  Requester ID:                      0x839b
```

Any ideas as to why me_cleaner isn't working? Given that the ME shows up in the flash region 2, I wouldn't think there would be a different chip I'd need to pull the firmware off of.

Thanks in advance.

precurse, 2018-10-14:
Thanks for the prompt reply!

So I'm assuming that means the ME section is stored on another chip somewhere on the board?

corna, 2018-10-15:
Could be.
Also make sure the firmware is not split between two chips.

precurse, 2018-10-17:
Found the second chip. It was unfortunately hidden nearby the first, but couldn't see or get at it without removing the case.

Getting the following now:
```
Full image detected
Found FPT header at 0x3010
Found 28 partitions
Found FTPR header: FTPR partition spans from 0x160000 to 0x210000
ME/TXT firmware version 9.1.42.3002 (generation 2)
Public key match: Intel ME, firmware version 9.0.x.x, 9.1.x.x
The AltMeDisable bit it NOT SET
Checking the FTPR RSA signature... VALID
```

Looks like this should. Thanks for the help.

Cheers

closed 2018-10-17 by precurse