corna/me_cleaner #238: HP Z200 -

lwoody7110, 2018-11-21:
Hi project team

I'm using this tool in the hope of resolving a 30min shutdown on 2x HP Z200 workstation motherboards purchased from eBay.

**me-cleaner --check original.rom** throws a WARNING

> Unknown public key 645f093c24ec0302df950e6dd1a51789
> Assuming Intel ME
> Please report this warning to the project's maintainer!

Please see photo of output in case of typo...
[image]

lwoody7110, 2018-11-24:
as a follow up, I'm using the guide [here](https://wiki.gentoo.org/wiki/Sakaki%27s_EFI_Install_Guide/Disabling_the_Intel_Management_Engine#Hardware_Configuration) 

**me_cleaner --soft-disable original.rom --output modified.rom** resulted in what appeared to be successful 'clean' but once applied did not disable the 30 minute timer.  Are you able to assist?

lwoody7110, 2018-11-26:
Is this project no longer active?

corna, 2018-11-27:
Still active, unfortunately I don't have much spare time anymore.

You can safely ignore that warning, it just means that the public key of Intel ME is not known by me_cleaner (see [these](https://github.com/corna/me_cleaner/blob/master/me_cleaner.py#L35-L53)).

lwoody7110, 2018-11-28:
Thanks for the reply.  me-cleaner seemed to have wiped out the BIOS of ME related parts but the motherboard still turns off after 30 minutes.  Is there anything else worth trying or shall I abandon this board to the skip?