corna/me_cleaner #242: New partition format on Gemini Lake?

Velsoth, 2018-12-12:
- [Intel Celeron N4000 processor](https://ark.intel.com/products/128988/Intel-Celeron-N4000-Processor-4M-Cache-up-to-2-60-GHz-)
- Gemini Lake
- [ASUS PN40 Mini PC](https://www.asus.com/us/Mini-PCs/Mini-PC-PN40/overview/)
- OEM BIOS
- Winbond W25Q128FW (1.8V)
- me_cleaner commit 43612a630c79f3bc6f2653bfe90dfe0b7b137e08
- flashrom on a Raspberry Pi 1 B+ with a TL866II 1.8V level converter

[image]

I tried my luck after reading @sakaki- 's [success report on the previous processor family (Braswell)](https://github.com/corna/me_cleaner/i=100x20ssues/3#issuecomment-434822048). Unfortunately it did not work on Gemini Lake:

```bash
$ python me_cleaner/me_cleaner.py -c asuspn40_1ceabeca8233659c202b278b97931698.img
Full image detected
The ME region in this image has already been disabled
The HAP bit is NOT SET
Reading partitions list...
Traceback (most recent call last):
  File "me_cleaner/me_cleaner.py", line 772, in <module>
    for i in range(entries):
NameError: name 'entries' is not defined
```

Sounds like the partition format has changed? I will give it a deeper look but I am opening this issue and attaching the ROM in case somebody can give it a go: [asuspn40_1ceabeca8233659c202b278b97931698.img.zip](https://github.com/corna/me_cleaner/files/2672605/asuspn40_1ceabeca8233659c202b278b97931698.img.zip)

On Intel ARK's page for this processor there are features that seem to rely on the existence of ME or TXE (Identity Protection Technology, Software Guard Extensions, OS Guard).

ajuiJM28, 2019-01-01:
Same for Asrock J3355M (Apollo Lake) Bios 1.90
https://asrock.com/mb/Intel/J3355M/?cat=Download&os=BIOS

and
Asrock J4005M (Gemini Lake) Bios 1.30
https://asrock.com/mb/Intel/J4005M/?cat=Download&os=BIOS#BIOS

Seems me_cleaner is getting difficulties to process TXE-like IME.

I think me_cleaner should support these boards as many ppl are using them for pfSense and NAS.

skochinsky, 2019-01-01:
these platforms likely use IFWI layout, see #175

shamen123, 2019-01-03:
ifdtool (coreboot/util/ifdtool from coreboot) has the ability to set the HAP bit for Gemini Lake BIOS. Dump with your SPI reader, run ifdtool on the BIOS with the -M 1 switch

ajuiJM28, 2019-01-21:
Thanks for the info, shamen123. It's definitively interesting solution!, although I would prefer to have full me_cleaner support so all the partitions are removed/zeroed for extra security.

thepcwiz101, 2019-08-17:
My hp laptop with the intel pentium n4000 has a tpm in it that can be disabled in bios. I recommend disabling all security features in the bios if you are wanting to do some research.