corna/me_cleaner #244: ASRock Z390M-ITX/AC: firmware is corrupted or missing

dtbartle, 2019-01-01:
When running "./me_cleaner.py --check <bios>" I get:
```
Full image detected
The ME/TXE region is valid but the firmware is corrupted or missing
```

I tried this on the publicly available BIOS files (1.10 and 1.20) as well as BIOS files dumped from my motherboard (which actually differ a fair bit from the "stock" BIOS files for some reason).

https://www.asrock.com/MB/Intel/Z390M-ITXac/index.asp#BIOS

dtbartle, 2019-01-01:
It looks like for whatever reason, me_cleaner ends up getting the ME region wrong.  I can find `$FPT` in the file (at 0x4000), while me_cleaner thinks it's at 0x3000.

platomav, 2019-01-01:
CSME 12 is not supported by me_cleaner as the CNP CSE region of the SPI chip uses a new structure called [CSE Layout Table](https://github.com/platomav/MEAnalyzer/blob/9cb0768fe0dc89e07ddfa09b8c6c1d82c06c221c/MEA.py#L323) which combines [IFWI v1.6 (BPDT v1)](https://github.com/platomav/MEAnalyzer/blob/9cb0768fe0dc89e07ddfa09b8c6c1d82c06c221c/MEA.py#L3017) for Code and [$FPT v1.0](https://github.com/platomav/MEAnalyzer/blob/9cb0768fe0dc89e07ddfa09b8c6c1d82c06c221c/MEA.py#L158) for Data/FS partitions.

dtbartle, 2019-01-04:
Yes.  I also tested against stock BIOS files.

dt-zero, 2019-06-14:
Would you mind testing https://github.com/corna/me_cleaner/pull/282 ?
Soft disable might work as expected. I'm having success with a similar board.

dtbartle, 2019-12-16:
Yes, confirmed your PR works for this board.