corna/me_cleaner #248: Supermicro X9SCM-iiF report

c-e-p-x-u-o, 2019-01-14:
I've just (looks like sucessfully) "cleaned" ME.

But intelmetool reports different results than specified  on https://github.com/corna/me_cleaner/wiki/Get-the-status-of-Intel-ME

exactly:
```
ME: Firmware Init Complete: YES
Progress Phase State: 0x1d
```

motherboard: Supermicro X9SCM-iiF
https://www.supermicro.com/products/motherboard/Xeon/C202_C204/X9SCM-iiF.cfm

me_cleaner output:
```
% python3 me_cleaner/me_cleaner.py -S -O x9scmii5.220_S-clean x9scmii5_220/x9scmii5.220          
Full image detected
Found FPT header at 0x30010
Found 5 partition(s)
Found FTPR header: FTPR partition spans from 0x60000 to 0x80000
ME/TXE firmware version 2.2.0.49 (generation 2)
Public key match: Intel SPS, firmware versions 2.x.x.x
The AltMeDisable bit is NOT SET
Reading partitions list...
 MFSB (0x00010000 - 0x000060000, 0x00050000 total bytes): removed
 FTPR (0x00060000 - 0x000080000, 0x00020000 total bytes): NOT removed
 MFS  (0x00080000 - 0x0000d0000, 0x00050000 total bytes): removed
 OPR1 (0x000d0000 - 0x0001d0000, 0x00100000 total bytes): removed
 OPR2 (0x001d0000 - 0x0002d0000, 0x00100000 total bytes): removed
Removing partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0x13)...
Reading FTPR modules list...
 BUP              (uncomp., 0x06035c - 0x07535c       ): NOT removed, essential
The ME minimum size should be 499712 bytes (0x7a000 bytes)
The ME region can be reduced up to:
 00030000:000a9fff me
Setting the AltMeDisable bit in PCHSTRP10 to disable Intel ME...
Checking the FTPR RSA signature... VALID
Done! Good luck!
```

intelmetool -m before:
```
Bad news, you have a `C204 Chipset LPC Controller` so you have ME hardware on board and you can't control or disable it, continuing...

MEI found: [8086:1c3a] 6 Series/C200 Series Chipset Family MEI Controller #1

ME Status   : 0xf0345
ME Status 2 : 0x4000c301

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : YES
ME: Manufacturing Mode      : NO
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Normal
ME: Current Operation State : M0 without UMA
ME: Current Operation Mode  : (null)
ME: Error Code              : No Error
ME: Progress Phase          : Module Loading
ME: Power Management Event  : Clean Moff->Mx wake
ME: Progress Phase State    : Unknown 0x00

ME: Extend Register not valid

Error mapping physical memory 0x0000000804270080 [0x2000] ERRNO=1 Operation not permitted
Could not map ME setup memory.
Do you have kernel cmdline argument 'iomem=relaxed' set ?
```

intelmetool -m after:
```
Bad news, you have a `C204 Chipset LPC Controller` so you have ME hardware on board and you can't control or disable it, continuing...

MEI found: [8086:1c3a] 6 Series/C200 Series Chipset Family MEI Controller #1

ME Status   : 0x21f0382
ME Status 2 : 0x101d9000

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : YES
ME: Manufacturing Mode      : NO
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Recovery
ME: Current Operation State : Bring up
ME: Current Operation Mode  : (null)
ME: Error Code              : No Error
ME: Progress Phase          : BUP Phase
ME: Power Management Event  : Clean Moff->Mx wake
ME: Progress Phase State    : 0x1d

ME: Extend Register not valid

Error mapping physical memory 0x0000000804265984 [0x2000] ERRNO=1 Operation not permitted
Could not map ME setup memory.
Do you have kernel cmdline argument 'iomem=relaxed' set ?
```

Yes, iomem=relaxed' is set, debian sid 4.19.0-1-amd64.

corna, 2019-01-14:
SPS is a bit different from ME, so some differences are expected. In this image, the FTPR partition contains just the BUP module, so the whole partition is kept untouched, which means that SPS can successfully complete the HW init.

The interesting part in your intelmetool output is `ME: Current Operation State : Bring up`, which tells us that SPS is stuck at the bring up.

closed 2019-01-14 by corna

c-e-p-x-u-o, 2019-01-14:
Thank you for explanation.