corna/me_cleaner #258: [Question] Extent and consequences of (successfully) disabling ME in modern systems.

Unb9rn, 2019-02-13:
Hello! I've tried googling a bit but there are little answers.
First, if I understand correctly, setting HAP bit disables ME very early, right after BUP phase. So no other module gets loaded. Now, if I recall correctly, many subsystems rely on ME(HECI), like firmware TPM, Integrated Sensor Solution and, probably SecureBoot (which may rely on TPM). So, if I disable ME successfully on a modern laptop, wouldn't I just cripple its' functionality? I am really eager to disable ME 11.8 on my new x360 Spectre, which does not have discrete TPM chip, but Secure Boot with custom PK/KEK seems to be much more useful for security reasons than disabled ME. Also, if I am killing gyro, accelerometer and other sensors, it kills most of the charm of this machine...
It would be great if anyone has info on what systems may end up disabled. Thanks!

skochinsky, 2019-02-13:
browse open issues to get a feela about possible consequences.
For example, there are some reports about sensors not working anymore, e.g. #161

weareanomalouswearearegion, 2019-05-28:
Secure Boot does not _need_ a TPM to function. For my usage, I need to disable Secure Boot anyway in order to use the NVMe mod to allow NVMe booting on X79 motherboards. Not to say, there's no TPM header on my X79 motherboard.

How it affect you may vary depending on exact hardware. For my Asus X79 motherboard the temperature sensors are still running as far as I can tell.