corna/me_cleaner #259: question - intel s5520hc can ME disable onboard NICs?

duecedriver, 2019-02-14:
I purchased a used server .. likely amazon/google.. a rackable systems based on intel s5520hc motherboard

both NICs light up but only one shows up in bios as having a mac address and usable for internet traffic.. both can get access to the BMC/ME so the ports physically work.

is it possible that the intel ME was used to disable a NIC port on the motherboard?  and if so.. can any ME too or bios screen be used to bring it back?

Thanks

skochinsky, 2019-02-14:
Are you sure both ports are actually connected to the main CPU/PCH? Probably one port is dedicated BMC port and is not accessible from the host OS.

duecedriver, 2019-02-14:
the board should have 2 full nics and both should also pass through to the BMC/ME.

the stickers show 3 mac addresses which match up with what the manual says.. but it appears 1 nic was disabled for 'security' according to the place I bought it from .. but could give no other details

kawalkowski, 2022-01-18:
I am pretty sure ME will not but the vendor (or you ) can configure this in the EEPROM of the 82575.  I had the same issue on my S5520HC from Rackable systems (SGI). You can enable it via eeupdate tool from intel. There is a built in hex editor into the tool if you launch it via eeupdate.exe /gui   (I used version 5.24 of the tool that i found on the internet)
You need to make sure the first 6 bytes are the correct address which needs to match the sticker on the NIC port.   My board was setup to the 2nd address of the 2 so just one symbol needed to change.  There is only one address you enter in the first 6 bytes the second NIC automatically get +1 address. 
Per 82575 manual, the location of interest is 10h is where the vendor can disable the port for NIC 2 and location 20h for NIC 1. Since mine was disabled on NIC 2 i copied 2 bytes from Location 20h to 10h to copy the settings which per manual take 16 bits. 
Its easy to do in the eeupdate utility  and just escape out of editing and the program will ask if you want to save to EEPROM.  

After that  I have both ports working.
[image]

duecedriver, 2022-01-18:
Holy necro thread revival batman!!!

@kawalkQCWski thanks for the tip

I am currently running esxi on that box with just the one (right) NIC working fully and the (Left) providing some BMC temp monitoring but it does not have a address... I will give this a try

can this be run on a 'live' usb key or does it need a windows session?  I cant pass the nic to windows in ESXI or I will loose control of the box

thoughts?

kawalkQCWski, 2022-01-19:
I ran it off bootable free dos disk made with rufus. 
Just insert the USB with EEPUDATE on it . F6 gets you to boot options . Pick your USB drive.

kawalkQCWski, 2022-01-19:
I wanted to  try port bonding to increase connection bandwidth and do not have room in my server for an add on card.  I have a few dual port and quad port cards but they will not fit without removing the Raid adapter (OMV box for me).  So  a little reading and a few failures later we have a solution...... Been running  24h no issues with both ports working and having individual MAC addresses.  

Got the needed file from  https://www.dropbox.com/s/u0hsje73rl7v6w7/ChangeMac.zip 
I used the version in the INTXL710 folder which is version 5.24

kawalkQCWski, 2022-01-19:
Here is the EEprom map  for reference ( loading before I lose this stuff)
[image]

http://notes-application.abcelectronique.com/027/27-45709.pdf

kawalkQCWski, 2022-01-19:
Seems it worked by accident as bit 11 in word 20H  is zero

duecedriver, 2022-01-19:
thanks for the addl info... I didn't realize that freedos could do GUI apps, always assumed it was command line

I dont know if my ports were disable the same way as yours... 

both ports I believe work for BMC but not ethernet.

only my right port looking at the backplane gets a address, and I dont recall if its the base address or the +1 its been a couple years before I gave up.

I will look into this and see what I can make of it... hex editing is not a strong suit of mine but I will try and figure it out..

thanks again for sharing!!

duecedriver, 2022-01-19:
I was looking at the manual to see if there is a 'reset to default' for eeupdate but am not finding one...

I will bring the esxi box down later and give this a shot

so 
create freedos with Rufus
move the files to the usb
boot the usb in non-uefi mode
start eeupdate.exe /gui

undo the disable bit
reset Mac address to the starting address (probably -1)
second address and bmc/RRM me addresses all get recalculated off of the base Mac +2 and +3

kawalkQCWski, 2022-01-19:
Yes all correct . The GUI operates only in hex  so most likely you will also see E30C in location  20h and E70C in location 10h. Intel counts these as 16 bit locations BTW for whatever reason. 

You can also make a backup of the EEPROM  via eeupdate.exe /dump. 
All instructions are displayed up via eeupdate -h if you want to do some other stuff.

duecedriver, 2022-01-19:
I think you are right that copying 20 to 10 worked by accident because I don't think the bit layout of the 2 addresses control the same functions, just by chance that changing only that one bit that controls lan disable makes the word hex value the same...

kawalkQCWski, 2022-01-19:
They actually control the same functions just bits 10,11,12  for LAN0 don't do anything so are set to 000 .All other bits control the same stuff.  Intel never wanted both ports disabled.

kawalkQCWski, 2022-01-19:
[image]

kawalkQCWski, 2022-01-24:
Did it work for you?

duecedriver, 2022-01-24:
@kawalkQCWski 

yep. Replied to you over at servethehome. 

Changed word 10. Rolled back the starting MAC by one and rebooted. Only strangeness is that nic2 on the right retained the correct MAC+1 address but became nic1 in bios and the left mic became nic2 even though it has the lower MAC address.  But seems to work fine. 

Thanks again for sharing!

kawalkQCWski, 2022-01-24:
Just looked over there. Glad its working.

wulfgar93, 2025-01-26:
@kawalkQCWski just wanted to thank you! Got the same issue today and you saved me a lot of time!

duecedriver, 2025-01-26:
its good to see that people are still rocking the s5520hc..  I still have my dual L5640 box up as its been hard pressed to find a cost effective path forward that had the power per watt per dollar upgrade.. 

I am messing with a dual E5-2630L v4 the I just picked up sub 300 bucks.. and I think at this price point .. the S5520 platform has met its match.. probably double the compute and 20-40 % power savings.. so far.. looks promising ..