corna/me_cleaner #264: Shrunken Intel ME causes OpenBSD to not detect NIC or HDD on the x220 Thinkpad

isthisme, 2019-03-17:
Hi.  I was attempting to install OpenBSD on my x220 Thinkpad with coreboot + the Intel ME stripped, AltMeDisable set, and the ME shrinked.  However I ran into two problems during the install that don't allow me to proceed with the installation:
- The ethernet device isn't detected causing OpenBSD to be unable to connect to the internet
- The HDD isn't detected either so OpenBSD can't even be installed, OpenBSD just states that there's no available disks

I would attach a dmesg but I have yet to find a way to upload it as a result of the issues I've outlined.

gch1p, 2019-08-22:
That's strange, I installed OpenBSD on corebooted X220 with shrinked ME just a few days ago and everything works.

elagost, 2019-11-09:
I too am having this issue. Have had my x220 corebooted for a long time and runs GNU/Linux with no issues but upon booting OpenBSD 6.6 flash drive it does not show any disks or the Ethernet adapter. Not sure if this is supposed to be reported here or not, but this is the only thing I could find with someone else having this same issue. @gch1p could you post your defconfig if yours is working?

gch1p, 2019-11-09:
@elagost can't do it now, but in a couple days i'll flash coreboot on my x220 and see.

elagost, 2019-11-09:
@gch1p  Wow, thanks for the super-speedy reply! I really appreciate it.

gch1p, 2019-11-14:
Ok this is weird. I just flashed coreboot and now OpenBSD can't see ethernet and HDD. I didn't touch ME.

But It was working in August. I'll ask on irc, maybe someone knows something.

elagost, 2019-11-14:
Huh, strange. I've got OpenBSD installed on a drive in my x230, and I'll see if I can swap that to the x220 and troubleshoot more there. The hardware is similar enough to where it should work.

0xsaso, 2021-02-16:
I know I'm resurrecting a pretty old issue here, but this is the only place I can find where someone has had the same problem as me. No matter what I do, the latest version of Coreboot will only give me two SATA modes to choose from:

1. AHCI
2. Compatible

The OpenBSD installer will only see my SSD if I choose "Compatible", but even still it doesn't see my NIC for ethernet. Did anyone in this thread have any success? I've tried with some older versions of the OpenBSD installer (as far back as 6.4) with the same problem and am thinking of trying older versions of Coreboot.

elagost, 2021-02-16:
I'll try and recompile coreboot on my x220 later and see if I can do the "Compatible" thing and get it to see the disk.
Also, realized I hadn't updated my last post - the swapping in the drive from the x230 didn't work. It didn't boot fully.

0xsaso, 2021-02-16:
> I'll try and recompile coreboot on my x220 later and see if I can do the "Compatible" thing and get it to see the disk. Thanks for the tip.
Interested in how it goes for you. I guess the only thing I haven't done is recompile without using me_cleaner to strip the bins, but doing so would sort of defeat the purpose.

elagost, 2021-02-17:
Confirmed that setting sata_mode to "Compatible" in nvramcui does allow the disk to show up in OpenBSD 6.8's x64 installer. Thanks for the tip!

I do get a message in the boot processes that says "the EEPROM checksum is not valid" and "em0: unable to initialize the hardware"

0xsaso, 2021-02-17:
I have that same error. Going to try and flash without stripping the firmware tonight and see if I get the same error. I also don't typically use a layout file - maybe that's part of my problem? I don't want to have accidentally touche the EC like this guy, but I also don't want to only flash a single region. https://mail.coreboot.org/hyperkitty/list/coreboot@coreboot.org/message/VQHJ6QFY2ZPEV3TUTC3P2R5WIAYKUYPA/

elagost, 2021-02-17:
I recompiled and flashed the latest this morning using existing blobs. Same errors.
After reading that thread it seems clear it's an OpenBSD bug, or at least not a bug with Coreboot or me_cleaner, since things do work fine under Linux and others. Wireless works just fine in OpenBSD though.

0xsaso, 2021-02-17:
If you install it, let me know if you're able to fix the ethernet issue afterward or anything. Would love to use it since it's my favorite OS, but not at the expense of a neutered Coreboot. The irony...

elagost, 2021-02-17:
I swapped my x230's disk in. It's an updated 6.8 install and has intel_firmware installed, so it gave this error:
`em0 at pci0 dev 25 function 0 "Intel 82579LM" rev 0x05: cannot find mem space`
Not sure if that helps any more. I'm probably just going to leave Linux on it, given that everything works there.