corna/me_cleaner #265: Not Working: ASRock H77M (Ivy Bridge/Intel H77 chipset)

ReaperX, 2019-03-21:
OEM site: https://www.asrock.com/mb/intel/h77m/

Ran me_cleaner on the latest (1.70) firmware with the -S option:
```
Full image detected
Found FPT header at 0x1010
Found 15 partition(s)
Found FTPR header: FTPR partition spans from 0x93000 to 0x108000
ME/TXE firmware version 8.1.20.1336 (generation 2)
Public key match: Intel ME, firmware versions 7.x.x.x, 8.x.x.x
The AltMeDisable bit is NOT SET
Reading partitions list...
???? (0x000003c0 - 0x000000400, 0x00000040 total bytes): removed
 FOVD (0x00000400 - 0x000001000, 0x00000c00 total bytes): removed
 MDES (0x00001000 - 0x000002000, 0x00001000 total bytes): removed
 FCRS (0x00002000 - 0x000003000, 0x00001000 total bytes): removed
 EFFS (0x00003000 - 0x00004b000, 0x00048000 total bytes): removed
 NVCL (NVRAM partition, no data, 0x00010511 total bytes): nothing to remove
 NVCP (NVRAM partition, no data, 0x0000a553 total bytes): nothing to remove
 NVJC (NVRAM partition, no data, 0x00004000 total bytes): nothing to remove
 NVKR (NVRAM partition, no data, 0x0001257d total bytes): nothing to remove
 NVSH (NVRAM partition, no data, 0x00007609 total bytes): nothing to remove
 NVTD (NVRAM partition, no data, 0x00001eac total bytes): nothing to remove
 GLUT (0x0004b000 - 0x00004d000, 0x00002000 total bytes): removed
 MDMV (0x0004d000 - 0x000093000, 0x00046000 total bytes): removed
 FTPR (0x00093000 - 0x000108000, 0x00075000 total bytes): NOT removed
 NFTP (0x00108000 - 0x00017d000, 0x00075000 total bytes): removed
Removing partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0xa3)...
Reading FTPR modules list...
 UPDATE           (LZMA   , 0x0df68a - 0x0df848       ): removed
 ROMP             (Huffman, fragmented data, ~2 KiB   ): NOT removed, essential
 BUP              (Huffman, fragmented data, ~56 KiB  ): NOT removed, essential
 KERNEL           (Huffman, fragmented data, ~135 KiB ): removed
 POLICY           (Huffman, fragmented data, ~91 KiB  ): removed
 HOSTCOMM         (LZMA   , 0x0df848 - 0x0e662b       ): removed
 RSA              (LZMA   , 0x0e662b - 0x0eb891       ): removed
 CLS              (LZMA   , 0x0eb891 - 0x0f1034       ): removed
 TDT              (LZMA   , 0x0f1034 - 0x0f7714       ): removed
 FTCS             (Huffman, fragmented data, ~18 KiB  ): removed
 ClsPriv          (LZMA   , 0x0f7714 - 0x0f7af4       ): removed
 SESSMGR          (LZMA   , 0x0f7af4 - 0x1063f1       ): removed
The ME minimum size should be 696320 bytes (0xaa000 bytes)
The ME region can be reduced up to:
 00001000:000aafff me
Setting the AltMeDisable bit in PCHSTRP10 to disable Intel ME...
Checking the FTPR RSA signature... VALID
Done! Good luck!
```
Flashed internally. The flash process (looked like it) succeed. At the reboot msg, I turned the machine off, removed the battery, waited, put the battery back in, powered up. UEFI works normally, except that it no longer remembered the contents of the EFI partition, so I booted from a USB stick and searched for EFI bootladers. My Manjaro Linux system booted normally from there.

Unfortunately, intelmetool indicates that ME is still active: 
ME: Firmware Init Complete  : YES
ME: Progress Phase: Host Communication
ME: Progress Phase State: Host communication established

I repeated the experiment with the -s switch, and then no flag, and I also tried with the previous bios version 1.6. Same outcome. :(

Diamondinthedirt, 2019-04-06:
@ReaperX as far as i understand these are mandatory modules to have the entire firmware removed from the system
 ROMP             (Huffman, fragmented data, ~2 KiB   ): NOT removed, essential
 BUP              (Huffman, fragmented data, ~56 KiB  ): NOT removed, essential

if these still there nothing else can be done except reflash on programmer

weareanomalouswearearegion, 2019-05-28:
> Flashed internally
Found your problem. I bricked my Rampage motherboard trying to flash internally using FPT, and Asus USB BIOS flashback does not touch the ME_region from my experience. You are better off buying an SPI flasher for a few bucks to flash it.

If you use an SPI flasher, it should most likely work.