corna/me_cleaner #266: me_cleaner.py", line 548, in <module>

Diamondinthedirt, 2019-03-24:
Hey Wassup,
I  did almost everything but following error occurs

C:\WINDOWS\system32>C:\Python37\python C:\me_cleaner-master\me_cleaner.py -S -O modified_image.bin original_dump.bin
Traceback (most recent call last):
  File "C:\me_cleaner-master\me_cleaner.py", line 548, in <module>
    f = open(args.file, "rb" if args.check or args.output else "r+b")
FileNotFoundError: [Errno 2] No such file or directory: 'original_dump.bin'

whats wrong i dunno, pls help me

skochinsky, 2019-04-04:
you need to either specify the full path to the file, or run the script while in `C:\me_cleaner-master`, not in `C:\WINDOWS\system32`.

Diamondinthedirt, 2019-04-05:
@skochinsky error "unknown  image"
[
[image]
](url)

Diamondinthedirt, 2019-04-05:
@skochinsky 
what next [??](url
[image]
)

Diamondinthedirt, 2019-04-05:
@skochinsky looks like it worked, but not sure
[image]

Diamondinthedirt, 2019-04-05:
... [image]

closed 2020-01-06 by truehomie