corna/me_cleaner #267: Interesting issue, Dell bios updating on Latitude 7450

czeej, 2019-03-26:
The dell bios managed to update remotely, I had disabled the Network stack on intel ME.
It wouldn't boot after untill I re-seated the ram. 

I may re image it and re flash ME. I feel replacing the Wifi card to one that does not require non-free firmware would maybe prevent this issue in the future untill I manually flash my Bios. 

It was A02 and had changed to A04. I know it's an old version but I was surprised to say the least the Bios updated.

Anyone else using dell notice the same thing? Cleaning ME can lead to bios updates, and the 32 bit bios isn't readable with flashrom. I think Dell would not be happy if someone managed to decompile it.

benjamindoron, 2019-07-22:
If it updated from A02 to A04, this might simply be LVFS - Linux Vendor Firmware Service (https://fwupd.org/).

czeej, 2020-03-08:
I have my doubts, I was only using free software and did not add in non-free repos. 

This happened and I could not boot screen would be buggy ext. Only got a boot after re-seating ram.  After I re-seated ram, bios was updated. It was near similar E7450 laptops as well Although I have no idea what happened.

mostav02, 2020-03-21:
It's somehow scary after reading your post. I had already other weird behaviors on this laptop, but well, will not going to describe them now because it may be unconfirmed paranoid stuff.

Is there any possibility you can try to provide more details on the circumstances when BIOS was automatically updated or you could do any further investigation on what happened? I have never experienced this issue, however I have a newer BIOS version.

I [googled](https://www.google.com/?q=dell+bios+automatic+update) and didn't find anything that mention automatic BIOS updates without user consent on when running Linux without UEFI. However, apparently there are some users who report automatic updates when on Windows 10, but I think it's only Win10 related.

czeej, 2020-03-22:
Yes, maybe it's only scary from a dell perspective but reading around the internet there doesn't seem to be a way for it to happen remotely besides windows10 that I know of. Just never had it update before I flashed it. Maybe totally un-related though.

Yes, it did update.. However, I probably dual booting at the time to run Inteli CAD at school. I guess that slipped my memory! I don't think there really is much to worry about. I think your right about windows doing the updating.

 If I remember maybe the failed booting I set the laptop down a little fast on a hard counter and the ram might of been only half seated. So it booted but visuals were green tinted and would crash at slim login screen. Re-seating would fix it. I think that was just me being paranoid. ^_^ I do mess with windows to close it but I usually try not to break updates and turn off driver updates in GP.

Although, that's good you got a free wifi card :).  I feel dell bios does have capabilities. They are enterprise choice usually, 
Yeah, got my memory jogging forgot I was dual booting for CAD I had 30 gb or so partitioned for windows and CAD.  Sorry to spread paranoia. 

But the Absolute Software was enabled.. Computrace fueling my paranoia. But I had contacted them by email gave them my Service tag and they told me they would disable it about when I made this post a year ago. It's still enabled in bios, but I emailed them recently about two weeks ago. Plugged in a windows install wired to modem for 4 days. They told me it made no calls to them. So it re-assured me they did disable it. Although Bios is still greyed out. So I am less paranoid now after talking directly with the guys at Absolute . lspci shows no trace of ime, so that is good sign.

As for Dell, I'm sure bios is not all that bad, although I know on older bios they need DOS to update. And Mr.Robot didn't help with the E corp symbolism. Anyway. A free wifi card should harden your system. Less non-free blobs the better :v: . 

This topic maybe should be closed, However, I still have the issue of not being able to get a good dump of the image anymore :) which I have opened a second topic about. I think maybe it's ok to be little paranoid, but it's maybe OK to relax a bit as well. Please forgive my ranting, I guess it was probably a windows updating the bios. Although, Oddly enough A04 was much older then the most recent revision.

mostav02, 2020-03-24:
In your case, since you disclosed some details about your usage and there was Windows OS involved, I believe it was caused by Windows exactly. In my case, while it could be called paranoia, I still believe my machine was intruded by some agency without my consent. The only attack vector I could find in my case is IME, since it provides Full Network manageability with KVM functionality on its own. I use a diskless configuration on this laptop where the Linux OS loads from a removable media in a `toram` mode, thus all the system binaries are always read-only, so there is very low probability of a local infection. What actually happens from time to time is my private unread chats are being read without my consent. Sometimes I just leave the laptop while doing other stuff and when I get back to it there are new private messages that I never had read are marked as read... I work in the infosec area for a long time and used all my experience to inspect the system and haven't found any sign of malware. It could be a bug in the chat software but I doubt, since it's a known and stable FOSS program. Yes I could email you if you drop your email address.

czeej, 2020-03-25:
Hey, interesting my email is jsoip@tuta.io. I have had similar intrusions running a free operating system (Mostly minus the wifi driver) (My git hub account is registered to outlook which is not encrypted or anything so)
Installed on bare metal. I believe this latitude 7450 might be a nasty one. My host name from the start was quite suiting I believe 😉