corna/me_cleaner #270: Works on:

0181505050, 2019-04-09:
hi peterwankman, do you think your method will work on a MAXIMUS VIII HERO motherboard with the same processor?

Id really like to give this a go. im new to programming with the universal programmer. but im quick at learning. and can follow instructions. if your keen to help give me a shout, thanks, tom. 

Works on:

- i7-6700K (Skylake)
- ASUS Z170-K
- OEM BIOS
- TL866 Universal Programmer
- with and without -S option (without causes delay)

I used a cheap USB programmer I had used for earlier projects. It automatically detected the flash chip and read from and wrote to it perfectly.
The original BIOS was the latest release from ASUS (3085 from 25.05.2018) with the latest ME Patch (11.8.55.3510 from 27.08.2018)
BIOS now reports ME Version 0.0.0.0; Windows Device Manager reports the ME as not connected.
No issues so far.

_Originally posted by @peterwankman in https://github.com/corna/me_cleaner/issues/3#issuecomment-449930625_

peterwankman, 2019-04-09:
How would I know? It doesn't depend on the processor, but the flash chip.
Have a look at the chip and see if it's listed in the programmer's software.
Mine is a Winbond 25Q128FV in a DIP8 package, which is supported.
I can't say anything about any other chip, other than to get a spare one first, in case it doesn't work.

0181505050, 2019-04-09:
Ok mate. Theres more to it than I thaught. I have a Winbond 25Q128FVIQ 1552. I think i need to do more research. As from what you just said, Ill need quite a bit of instruction. cheers.

peterwankman, 2019-04-10:
https://www.winbond.com/resource-files/w25q128fv%20rev.m%2005132016%20kms.pdf
The -IQ at the end stands for the variant of the chip. It's the same as mine.
1552 is probably a date code (52nd week of 2015)

0181505050, 2019-04-10:
thanks peter. thats given me the confidence to do it. I need to research every step on the way. ( buy this tool you have) and learning how to backup, modify and reflash. I only want to do it once and do it right :) sadley no youtube videos or step by step guides for fools. ill get there eventually.

dartraiden, 2019-05-11:
> step by step guides for fools

https://github.com/corna/me_cleaner/issues/98

0181505050, 2019-05-11:
i am one grateful fool :)

Sent with [ProtonMail](https://protonmail.com) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

0181505050, 2019-05-11:
I assume the process is identical if you are doing it to a asus maximius viii?

Sent with [ProtonMail](https://protonmail.com) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

dartraiden, 2019-05-11:
Yes, I did it with Z170-P... but it is identical for any motherboard with a removable chip

0181505050, 2019-05-11:
I ordered the parts. i hope this works. really appreciate the effort thanks.

Sent with [ProtonMail](https://protonmail.com) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐