corna/me_cleaner #272: Not working on Lenovo Thinkpad T440p (i5-4300M, QM87 Express LPC)

dhaninugraha, 2019-05-06:
Hi Nicola,

I recently acquired a former business-unit Lenovo Thinkpad T440p with a borked ME -- evident by the unit abruptly shutting down every 30 minutes, MEInfoWin kept erroring out and attempts to install Intel Management Engine software always failed.

The machine has separate chips for BIOS and ME, so I dumped the ME image and applied me_cleaner, but still got the abrupt shutdown.

Removing CMOS battery did not help, as did `fptw -greset` followed by a CMOS battery removal.

Out of curiosity, I booted a live Ubuntu USB on the machine and ran `intelmetool -s`, and this is the output I got:
[image]

Does this mean I'm screwed? Should I just repair/clean the BIOS/ME image, reflash them and go on with my life?

Many thanks in advance.

Usernameisalreadytaken5, 2019-05-12:
Have same laptop t440p 

Juts download:
https://downloadcenter.intel.com/download/27150/INTEL-SA-00086-Detection-Tool

I got all bios update and i have:

**This system is not vulnerable. It has already been patched**

Read more:
https://www.ghacks.net/2017/11/22/find-out-if-your-intel-cpu-is-vulnerable-to-intel-manageability-engine-vulnerabilities/

[image]

moral-panic, 2019-05-12:
> Have same laptop t440p
> 
> Juts download:
> https://downloadcenter.intel.com/download/27150/INTEL-SA-00086-Detection-Tool
> 
> I got all bios update and i have:
> 
> **This system is not vulnerable. It has already been patched**

Not sure why/how this is relevant.

The INTEL-SA-00086 Detection Tool looks for SA-00086 related vulnerabilities and, if found, will in most cases, recommend a manufacturer's firmware upgrade that patches these vulnerabilities.    INTEL-SA-00086 was Intel's response to the outcry when ME became public knowledge.  It was the bone they threw to the world to placate and satisfy the public.  It does patch certain holes that need to be patched, yet it in no way does a patched firmware render a system "not vulnerable".  "less vulnerable" is more accurate.  The very nature of ME presents a grave and egregious security vulnerability that cannot be patched with firmware, hence the significance of me_cleaner.

Usernameisalreadytaken5, 2019-05-12:
I'm just use WIFI everywhere not a rj45 so i'm think is OK maybe 

Is your tools work and you have binary python to exe ?

maybe i will try it

corna, 2019-05-16:
Since it has the 30 minutes problem _before_ applying me_cleaner, it seems that the ME partition on your chip had been corrupted somehow previously. me_cleaner is unable to restore a broken partition.

Your best chance is to download a **clean** ME firmware image (like from [here](https://www.win-raid.com/t596f39-Intel-Management-Engine-Drivers-Firmware-amp-System-Tools.html)), dump the content of **both** the chips with flashrom, concatenate them, replace the broken ME firmware with `ifdtool -i me:valid_me_firmware.bin dump.bin` and flash back the result (split in the two parts).

corna, 2019-05-16:
If it works, then you can go with the usual me_cleaner procedure.

null-von-sushi, 2019-07-28:
Not sure if it helps you, but I have a T440p (the one with nvidia graphics, in case it matters (probably not)) and I took a backup of the original ME image for the T440p with a CH341A. I can upload them if you want. You could just flash that to (hopefully) make your ThinkPad work again, then run me_cleaner on it yourself?

KylesDigitalLab, 2021-12-16:
> Not sure if it helps you, but I have a T440p (the one with nvidia graphics, in case it matters (probably not)) and I took a backup of the original ME image for the T440p with a CH341A. I can upload [...]

null-von-sushi, 2021-12-17:
Sorry man, I do not :(

alphamaestro555, 2022-02-08:
Do you still need it?

> > Not sure if it helps you, but I have a T440p (the one with nvidia graphics, in case it matters (probably not)) and I took a backup of the original ME image for the T440p with a CH341A. I can upload them if you want. You could just flash that to (hopefully) make your ThinkPad work again, then run me_cleaner on it yourself?
> 
> Do you still have that, I need it.

Do you still need it?

alphamaestro555, 2022-02-09:
> Do you still have that, I need it.
Here ya go, if you still need it. Its from a T440p with dGPU.

SHA256sums:
883d6ec2645e388a95ca52e24e73ad872a56136875c59a25ba9ac7111b2afdce  original_dump4MB.bin
25ac0c4daa304cf61a95fb0534bfe6819672b8648e6b781b14a72f8ca9ca54ec  original_dump8MB.bin

[T440p_flash_images.zip](https://github.com/corna/me_cleaner/files/8036622/T440p_flash_images.zip)