corna/me_cleaner #273: Unknown images: ASUS P5K-E WiFi (ICH9R, from 2007)

R8s6, 2019-05-09:
Hi everyone,

The [libreboot docs](https://libreboot.org/faq.html) claim that "...the ME is present on all Intel desktop, mobile (laptop), and server systems since mid 2006."

This board, [ASUS P5K-E WiFi](https://www.asus.com/us/Motherboards/P5KEWIFIAP/overview/), is from 2007 so it should probably have it. The CPU is Intel Core 2 DUO E8400.

The file read from the SIP8 chip appears to be an encrypted/compressed file, so I had to use mmtool v3.26 to extract the uncompressed BIOS (named as "sip8_original.bin" below). 

Here's its ifdtool output:

``` 
./ifdtool -d sip8_original.bin
ICH Revision: 100 series Sunrise Point
FLMAP0:    0x661775f9
  NR:      6
  FRBA:    0x170
  NC:      2
  FCBA:    0xf90
FLMAP1:    0x67c0b70f
  ISL:     0x67
  FPSBA:   0xc00
  NM:      3
  FMBA:    0xf0
FLMAP2:    0x87892666
  PSL:     0x8926
  FMSBA:   0x660
FLUMAP1:   0x00000000
  Intel ME VSCC Table Length (VTL):        0
  Intel ME VSCC Table Base Address (VTBA): 0x000000

ME VSCC table:

OEM Section:
00: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
20: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
30: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Found Region Section
FLREG0:    0x00000000
  Flash Region 0 (Flash Descriptor): 00000000 - 00000fff 
FLREG1:    0x00040100
  Flash Region 1 (BIOS): 00100000 - 00004fff (unused)
FLREG2:    0x54534f50
  Flash Region 2 (Intel ME): 00f50000 - 00453fff (unused)
FLREG3:    0x53435f32
  Flash Region 3 (GbE): 00f32000 - 00343fff (unused)
FLREG4:    0x01004745
  Flash Region 4 (Platform Data): 00745000 - 00100fff (unused)

Found Component Section
FLCOMP     0x00000000
  Dual Output Fast Read Support:       not supported
  Read ID/Read Status Clock Frequency: 20MHz
  Write/Erase Clock Frequency:         20MHz
  Fast Read Clock Frequency:           20MHz
  Fast Read Support:                   not supported
  Read Clock Frequency:                20MHz
  Component 2 Density:                 512KB
  Component 1 Density:                 512KB
FLILL      0x00000000
  Invalid Instruction 3: 0x00
  Invalid Instruction 2: 0x00
  Invalid Instruction 1: 0x00
  Invalid Instruction 0: 0x00
FLPB       0x00000000
  Flash Partition Boundary Address: 0x000000

Found PCH Strap Section
PCHSTRP0:  0x00000000
PCHSTRP1:  0x00000000
PCHSTRP2:  0x00000000
PCHSTRP3:  0x00000000
PCHSTRP4:  0x00000000
PCHSTRP5:  0x00000000
PCHSTRP6:  0x00000000
PCHSTRP7:  0x00000000
PCHSTRP8:  0x00000000
PCHSTRP9:  0x00000000
PCHSTRP10: 0x00000000
PCHSTRP11: 0x00000000
PCHSTRP12: 0x00000000
PCHSTRP13: 0x00000000
PCHSTRP14: 0x00000000
PCHSTRP15: 0x00000000
PCHSTRP16: 0x00000000
PCHSTRP17: 0x00000000
AltMeDisable bit is not set

Found Master Section
FLMSTR1:   0x0000eff0 (Host CPU/BIOS)
  Platform Data Region Write Access: disabled
  GbE Region Write Access:           disabled
  Intel ME Region Write Access:      disabled
  Host CPU/BIOS Region Write Access: disabled
  Flash Descriptor Write Access:     disabled
  Platform Data Region Read Access:  disabled
  GbE Region Read Access:            disabled
  Intel ME Region Read Access:       disabled
  Host CPU/BIOS Region Read Access:  disabled
  Flash Descriptor Read Access:      disabled
  Requester ID:                      0xeff0

FLMSTR2:   0x000a8900 (Intel ME)
  Platform Data Region Write Access: disabled
  GbE Region Write Access:           disabled
  Intel ME Region Write Access:      disabled
  Host CPU/BIOS Region Write Access: disabled
  Flash Descriptor Write Access:     disabled
  Platform Data Region Read Access:  disabled
  GbE Region Read Access:            enabled
  Intel ME Region Read Access:       disabled
  Host CPU/BIOS Region Read Access:  enabled
  Flash Descriptor Read Access:      disabled
  Requester ID:                      0x8900

FLMSTR3:   0x800002c0 (GbE)
  Platform Data Region Write Access: disabled
  GbE Region Write Access:           disabled
  Intel ME Region Write Access:      disabled
  Host CPU/BIOS Region Write Access: disabled
  Flash Descriptor Write Access:     disabled
  Platform Data Region Read Access:  disabled
  GbE Region Read Access:            disabled
  Intel ME Region Read Access:       disabled
  Host CPU/BIOS Region Read Access:  disabled
  Flash Descriptor Read Access:      disabled
  Requester ID:                      0x02c0

Found Processor Strap Section
????:      0x06720000
????:      0x66f800b4
????:      0xb4f9c35e
????:      0xe8f8eb87
????:      0x00000392
????:      0x5666f1eb
????:      0xff01be66
????:      0x75fe8566
```
In order to make sure, I also downloaded [the official "BIOS" file](https://dlcdnets.asus.com/pub/ASUS/mb/socket775/P5K-E/P5K-E-1305.zip), and extracted its own uncompressed file. `diff` shows that this file from official website (after extraction) is the same as the one read from SIP8 (after extraction); The ifdtool outputs of the 2 files are also the same.

But, me_cleaner can't recognize it, here's me_cleaner output:
```
$ python me_cleaner.py -c sip8_extract.bin
Unknown image
```

Also trying the one from ASUS website:
```
$ python me_cleaner.py -c factory_extract.bin
Unknown image
```

Fired up the motherboard and used intelmetool:
```
$ sudo intelmetool
Not sure if ME hardware is present because you have a `82801IR (ICH9R) LPC Interface Controller`, but it is possible to disable it if you do, continuing...
RCBA at 0xfed1c000
MEI not hidden on PCI, checking if visible
MEI device not found, huh?
exiting
```

During flashing I found this interesting part on the motherboard:
[image]

Judged from the photo where there are 2 bright dots at the 1 and 2 positions, does it mean, if we trust ASUS (we probably shouldn't but let's pretend we do for now), the board is currently at the "1-2 Diable Me" configuration?

All in all, does this motherboard have ME enabled?

If so, how to remove it using me_cleaner or other tools?

Thank you very much!

moral-panic, 2019-05-10:
Did intelmetool actually output "exciting" ?

R8s6, 2019-05-10:
> Did intelmetool actually output "exciting" ?

You got me excited! I looked back and found it actually said "exiting". :D