corna/me_cleaner #274: I used Flashrom to brush the firmware made by me_cleaner on the Broadwell tablet, and then my tablet couldn't boot.

daiaji, 2019-05-15:
I used the `-S` parameter as an example.
`python me_cleaner.py -S -O modified_image.bin original_dump.bin`
And use flashrom to brush in the firmware.
`flashrom -p internal -w modified_image.bin`
My tablet is not working after rebooting.

So, no matter how I should use an external SPI programmer instead of using Flashrom to write firmware directly in the OS?

Does the `-S` parameter work on a Broadwell device?

Diamondinthedirt, 2019-05-16:
not properly composed request to provide any corrective action

weareanomalouswearearegion, 2019-05-28:
I suggest you to switch to a SPI flasher. I attempted to use FPT on my Rampage 4 and it bricked it. I only fixed it today using a CH341a flasher.

Diamondinthedirt, 2019-05-28:
@weareanomalouswearearegion CH341a is not a panacea, it is the cheapest programmer sold on the market, it may not work for all chips, modern chips has secure region, area with protection, that will not allow to overwrite the code correctly, these chips needs special programmers

Diamondinthedirt, 2019-05-28:
@weareanomalouswearearegion dont use CH341a on modern notebooks / laptops , never! i warned you..