corna/me_cleaner #276: Should i upgrade ME Firmware before me_cleaner ?

Denyreal, 2019-05-17:
I'm a hobbyist interested by removing management engine from the machine.

I would like to know if operate an upgrade of Management Engine firmware is sure because the manufacturer of the machine (HP) provide some updates to prevent several CVE.

The machine is shipped with [ i7-6700 CPU (Skylake) ](https://ark.intel.com/content/www/us/en/ark/products/88196/intel-core-i7-6700-processor-8m-cache-up-to-4-00-ghz.html)

Apparently HP provides some updates for :
* BIOS
* IME Firmware to 11.8.65.3590
* IME Driver 

Should i upgrade all these before removing ME ?

Any advice appreciated. 

Thanks in advance.

Diamondinthedirt, 2019-05-25:
lets get it straight, what do you want? delete or upgrade functionality.. ??

Denyreal, 2019-06-04:
Clearly dude ? I want to remove this shit !

truehomie, 2019-06-04:
@Denyreal so all you have to do further is to flash your modifyed BIOS image

cuvtixo, 2019-06-05:
It is good to upgrade all firmware first, but not necessary. In the title
you call it "ME firmware", so the question is confusing. I think you meant
"update all firmware"- if HP has updates, they are not going to give you a
choice about selectively updating the Management Engine. Also It seems like
you may be using a translation program. "Clearly dude!" Is a very unusual
English phrase. Be aware that translators can have limitations. You will
have to make an extra effort to be understood.

Denyreal, 2019-06-25:
Thanks guys for those advises.

In fact i do not use translator but yes english is not my native language and i meant by "clearly dude ?" "seriously ?"
Sorry for this . 

HP published upgrades of ME firmware for my Machine so i upgraded BIOS
and ME Firmware . 

I'll write notes on how i disabled ME if i succeed on my specific platform.