corna/me_cleaner #283: WORKS on Z390 Aorus Master

zutoxona, 2019-06-16:
Here to report that with the pull request from https://github.com/corna/me_cleaner/pull/282, me_cleaner worked for my Z390 Aorus Master with firmware v. F8. I needed to use the @bios utility (specifically https://www.gigabyte.com/MicroSite/121/tech_a_bios.htm), as trying to flash the modified firmware from UEFI returned an "Invalid Image" error.
Had to reset CMOS as i wasn't getting any video output, but after doing so everything seems to work correctly.

Thanks for all your work!

szungy, 2019-06-16:
Happy to hear your good results, thank you for sharing!

dt-zero, 2019-06-16:
Happy to hear your good results, thanks for sharing!

czeej, 2020-03-08:
I notice Gigabyte boards seem to have some of the easier to access Bios spi chips. Just from images of the Mobos . Could be wrong though.

schattenpanzer, 2020-06-12:
add it to number #3