corna/me_cleaner #284: Thinkpad e540 has dual bios chips

daviem, 2019-06-30:
Hi, I seriously need to disable intel management system, my Thinkpad has  and is being compromised, it has dual SOIC 8-PINS bios chips, that mean two bios chips,  how would I follow the external flash me_cleaner procedure?

Do I dump and reflash both chips seperately?

szbnwer, 2019-06-30:
hi there! :)

just out of curiosity, how do u know that its compromised? :O im really interested in every detail that u know!

thx for any info, all the bests to u! :)

daviem, 2019-07-01:
I launched a global business then my life started falling apart, false things occurring, Linux, nothing would stop accounts being hacked, false framing occurring, no firewall nothing, the router lights would constantly flash, even when off, I didn’t know what was going on until I found out about the intel engine so I want it secured.

Any ideas about the two bioses as theirs more computers in the house I’m sure are compromised and most contain dual bios Lenovo chips.

daviem, 2019-07-01:
Ps Hi :)

szbnwer, 2019-07-01:
thx, however this wasnt too clear to me :D so if you disable ur server app, network connection or even try a live linux, then you could see things that comes from nowhere? did u try wireshark? what if an application tries to call home? i think if u would have an nsa attack, then u wouldnt see things go mad, but things that trying to keep hidden... u can also try clamav, rkhunter and chkrootkit (search for their results, they can show false positive stuffs!) but a simple live media is somewhat more bulletproof, like i think tails wouldnt try to call home in every second :D otherwise i think that u could ask freenode/#coreboot or their mailing list, but there are massive guys, so try ur best when ull tell them ur case study, but i think they will b friendly, just say that ur really suspicious about ur stuff, but let them tell what u can try to make things sure :dagger: 

btw whats ur software stack?

otherwise im still just a spectator around here and around coreboot, so others will probably give more precise advices. good luck and come back if u have any news! :)

daviem, 2019-07-01:
Basically with all devices switched off the router would flash, at first I would find other computers attached to the router not mine, so I used an openwrt one I flashed and made highly secure no more unknown devices attached to the routers BUT in windows command lines would appear while the pc froze then a whole load of code would appear then disappear this is with Kaspersky on, with Linux the password would change, it would randomly freeze and unfreeze, with all devices disconnected and switched off the router still showed connections being made despite a robust iptables firewall. The same thing is still happening, Linux I used Ubuntu, mint, this is kinda big they are trying to destroy me so much bad has come my way and I haven’t done anything except start a business that was deemed to make millions now it’s in tatters.

szbnwer, 2019-07-01:
uhh, that sounds bad... did u consult with coreboot folks? i hope the bests to u!

off:
and thats why i still dont have my server. :D i wanna build it on the bare minimum, with a beagle bone, linux, lua and maybe some minor basic security tools only, and with every requests properly registered, just for taking away as much unknown factors as i just can... i hope i can rely on my security knowledge, but ill read some more rfc and owasp :D 

(btw that :dagger: wanted to be a :D in my previous message, just before u would think about it too much! :D )

daviem, 2019-07-01:
Thanks, I don’t know how to contact them, I just read about the 20 year cpu flaws too 😒, it looks like no matter what our computers/servers are full of hardware vulnerabilities.

It’s ok about the sword :))

szbnwer, 2019-07-01:
freenode irc: #coreboot or they have a mailing list, i dunno how to subscribe, ull find it on their website :D im on both places, but i check their irc rarely...

(and nope, thats a ": dagger :" just without spaces... ":D" makes it appear in a list and hitting <enter> will make it become a dagger... basically i care about not letting stuffs expand, but "discuss" forums wont give me such nonsense and that disabled my mental alarm... sad story)

szbnwer, 2019-07-01:
btw dont forget to bring back any valuable results, i had to realize that there are much ppl here who maybe even doesnt know about coreboot, but the me_cleaner folks will be able to close ur issue after any valuable info comes here about it...

szbnwer, 2019-07-01:
and u need to register a nick on irc if u wanna talk with them there...

here are my notes about that:

freenode reg:
/msg NickServ REGISTER password youremail@example.com
/msg NickServ VERIFY REGISTER yournick ************
hexchat:
ctrl+s
freenode->edit
sasl
pw

Stitch626, 2020-08-07:
Wow, this topic was kind of intense to read. However, dual-chip designs are usually just split images. So so you can basically put one image at the other. Depending of the design. Sometimes there is kind of container/overhead/whatever on those dual-chip things. I'd also recommend reading some coreboot stuff, there may be answers to you dual-chip question as well.