corna/me_cleaner #287: msi h81m-p33 | new overclocking options appeared after flashing, overclock settings not applying

mssx86, 2019-07-13:
1. 1. got the latest firmware, v.10 (E7817IMS.1A0), from msi support's downloads section
2. 1. ran `python3.7 me_cleaner.py -S -O E7817IMS.1A0.MODIFIED E7817IMS.1A0`
3. renamed `E7817IMS.1A0.MODIFIED` to `E7817IMS.1A0`
4. put it in the root of a fat32/vfat formatted usb drive
5. entered the bios, selected the bios/uefi flash utility
6. selected the modified firmware and flashed it using the `flash bios and me` option

now i have many more overclocking options like `BCLK` manipulation, which i didn't have before. problems:
1. i can't apply my previous overclocking settings, they disappear after a reboot.
2. trying to revert back to the unmodified stock firmware doesn't work. i still have those extra menu items under the overclocking menu and i've lost the ability to overclock completely.
3. as stated above, when i flash the stock firmware, it simply doesn't revert back. the `details` menu says `intel me version: n/a` no matter which version of bios firmware i'm at, even if on stock/unmodified ones.

here's the output of `intelmetool`:
```
Bad news, you have a `H81 Express LPC Controller` so you have ME hardware on board a
nd you can't control or disable it, continuing...

MEI found: [8086:8c3a] 8 Series/C220 Series Chipset Family MEI Controller #1

ME Status   : 0x1e020191
ME Status 2 : 0x1b4d2142

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : NO
ME: Manufacturing Mode      : YES
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Initializing
ME: Current Operation State : Bring up
ME: Current Operation Mode  : Debug
ME: Error Code              : No Error
ME: Progress Phase          : BUP Phase
ME: Power Management Event  : Power cycle reset through Moff
ME: Progress Phase State    : 0x4d

ME: Extend SHA-256: 393ffb341d635e1b11e5f5d155496bfed996e7435a367e564740b8bb4038796a

ME: failed to become ready
ME: failed to become ready
ME: GET FW VERSION message failed
```

what should i do to at least gain my ability to overclock and revert to the original state of my motherboard?

youiopmop, 2019-07-14:
using an external flasher like a ch341a or a raspberry pi with a soic-8 clip and flashing the stock firmware could work, but you would be missing stuff like MAC addresses for the ethernet card from the firmware if you use the downloaded copy from msi's site. maybe you could use a hex editor to fix it?

mssx86, 2019-07-14:
> using an external flasher like a ch341a or a raspberry pi with a soic-8 clip and flashing the stock firmware could work, but you would be missing stuff like MAC addresses for the ethernet card from [...]

i don't have any other machine than this, so am i just stuck with this broken bios unless i can find a rpi which is more expensive than the board itsellf?

dartraiden, 2019-07-14:
H81 chipset doesn't have overclocking abilities at all.

youiopmop, 2019-07-14:
> H81 chipset doesn't have overclocking abilities at all.

it does, but only for the pentium g3258, other K series CPUs aren't overclockable

dartraiden, 2019-07-14:
1) Unlock ME region, so you can overwrite it under Windows/Linux
https://www.win-raid.com/t3553f39-Guide-Unlock-Intel-Flash-Descriptor-Read-Write-Access-Permissions-for-SPI-Servicing.html
2) use FPT from Intel ME System Tools to flash new BIOS

dartraiden, 2019-07-14:
btw, if you have ME in working state (as intelmetool said), you should try to overwrite ME firmware (unlocking is not required if new version > or =)

1) obtain [Intel ME 9.0 1.5MB Firmware v9.0.31.1487](https://www.win-raid.com/t596f39-Intel-Management-Engine-Drivers-Firmware-amp-System-Tools.html) and Intel ME System Tools 9.1

2) `fwupdlcl64 -f mw.bin`

mssx86, 2019-07-14:
> 
> 
>     1. [Unlock ME region](https://www.win-raid.com/t3553f39-Guide-Unlock-Intel-Flash-Descriptor-Read-Write-Access-Permissions-for-SPI-Servicing.html) in some way (hidden BIOS setting or pinmod), so you can overwrite it under Windows/Linux
> 
>     2. [extract ROM from UEFI Capsule](https://www.youtube.com/watch?v=REOYyszYgF4) with [UEFITool 0.26.0](https://github.com/LongSoft/UEFITool/releases)
> 
>     3. use FPT from [Intel ME System Tools 9.1](https://www.win-raid.com/t596f39-Intel-Management-Engine-Drivers-Firmware-amp-System-Tools.html) to flash new BIOS
> 
> 
> `fptw64 -f new.rom`

dartraiden, thanks a bunch. i've tried the second method first and it gave me an error but the first method worked beautifully. now my bios is back to how it was without extra options and looks like me is as well.

thanks again.

closed 2019-07-14 by mssx86

dartraiden, 2019-07-14:
> i've tried the second method first and it gave me an error

This was to be expected because if the descriptor and the ME region are locked from writing (according to Intel recommendation, except for some Gigabyte's mobos) there is only one way to overwrite the ME region — ask ME to update itself. But, since ME was disabled, it failed.