corna/me_cleaner #289: x230i (i3) can't enter BIOS settings after applying me_cleaner

asvc, 2019-08-04:
Laptop works as expected, but whenever I try to enter BIOS settings it ends-up stuck showing the black screen.

I have tried <kbd>Ctrl+P</kbd> and <kbd>del</kbd> keys during startup - result is the same.

Worth noting, flashing with `-s` flag enables "Configure Intel Management Engine" menu item which, unfortunately, works as good as the BIOS settings themselves.

EEPROM dumps:
* [Unmodified top chip (4mb)](https://f001.backblazeb2.com/file/publicist/x230i/x230_MX25L3205-A_top.rom)
* [Bottom one (8mb)](https://f001.backblazeb2.com/file/publicist/x230i/x230_MX25L6405_bottom.rom)
* [Bottom one with me_cleaner applied](https://f001.backblazeb2.com/file/publicist/x230i/me_removed_with_descriptor.rom)

szbnwer, 2019-08-05:
hi there! :)

i think you should concatenate the 2 parts apply me_cleaner that way and then cut it and finally reflash the 2 new parts. this is just a guess to try, but ive seen this technique around here and i think it was in the near past, so look after it in the more recent issues or in #3 . otherwise thats all i can help, but others will know better if u need more help...

all the bests to u! :)

asvc, 2020-09-19:
@7XFugvwR , on the contrary - I love this bug, it effectively locks-down the BIOS settings which are inaccessible without a reflash.

7XFugvwR, 2020-09-21:
Ok, unfortunately i applied "default settings" before my flashing attempts that i wanted to revert after.

I figured out, that by using me_cleaner without "-S" option - just stripping down - it works again.

According to intelmetool the ME is disabled.