corna/me_cleaner #291: How do you unlock cpu descriptor settings?

evripidis23, 2019-08-12:
I try to update my bios, specifically the update giving me the ME8 which i need in order for my new gpu to work. My mobo is p67a-gd55 (ms-7681) I tried various methods to get this to work, including a tool called HQ flasher.
Then i get this message
The host cpu does not have write access to the target flash area: to enable write access for this operation you must modify the descriptor settings to give host access to this region
Using the cmos button did not help by the way

dartraiden, 2019-09-18:
https://www.win-raid.com/t3553f39-Guide-Unlock-Intel-Flash-Descriptor-Read-Write-Access-Permissions-for-SPI-Servicing.html

admin-elvistrujanovic, 2023-11-24:
What about Lenovo G580 InsydeH2O?