corna/me_cleaner #294: Lenovo X250 compatibility?

ginto37, 2019-09-07:
Someone has offered me a Lenovo X250 at a good price, but I wanted to see what success people had with it first. It comes with i7 CPU with vPro and a genuine, fully activated Windows 10 Enterprise license. According to *intelmetool* the controller is Wildcat Point-LP, which is supported by me_cleaner.  Still I know that that's not the full story (!) when it comes to compatibility, so anyone had any good experiences?

The bad thing is that Boot Guard Verified and Measured Boot is enabled, so it wouldn't accept Coreboot, but I would be happy as long as I can de-blob the ME /and/ enable the HAP bit.

Ph0rkoz, 2019-10-19:
You can't deblob the ME but you can set the hap and unlock the FPT. Bios will now complain it can't load the ME settings module when you enter it. ME confirmed initializing forever with coreboot utilities.

fincham, 2019-10-22:
Had success with external SPI programmer and the "-s" option passed to me_cleaner.py

ginto37, 2019-10-22:
Phorkoz:
> You can't deblob the ME but you can set the hap and unlock the FPT. 

Sorry, what's the FPT?

> Bios will now complain it can't load the ME settings module when you enter it. ME confirmed initializing forever with coreboot utilities.
> 
Are these just cosmetic issues? No security, usability, performance or
battery implications?

Michael Fincham:
> Had success with external SPI programmer and the "-s" option passed to
me_cleaner.py
>

That's good news. I'd prefer to de-blob, but neither you or Phorkoz
could get that to work so I'll have to be satisfied I suppose. *grin*

Just curious, how secure is setting the HAP bit? Should I be concerned
about the HAP bit being _unset_ again via some rogue or hidden mechanism?

Ph0rkoz, 2019-10-24:
Flash partition table. If you don't unlock that there should be no way for the bit to be overwritten. Maybe if you update bios.