corna/me_cleaner #295: What do the ME Capabilities mean, exactly?

ginto37, 2019-09-07:
Hi,

I wanted to make this a separate post even if it's about the same device. The X250 I posted about has a lot of the ME Capabilities listed as "ON" by *intelmetool*, and I decided I'd like to know what that means before buying it. 

The one I'm concerned about the most is the `Wireless LAN (WLAN)` option - does this mean I'd have to use a USB Wi-Fi dongle to be sure that the ME wasn't monitoring and using Internet traffic?  Which other ones should I be concerned about? 

I thought this thread might make a good reference for people, because there doesn't seem to be much about this, not that I could find after a good search even on Intel's website. Maybe others could post back with more info?

> ME Capability: Full Network manageability                 : ON
> ME Capability: Regular Network manageability              : OFF
> ME Capability: Manageability                              : ON
> ME Capability: Small business technology                  : OFF
> ME Capability: Level III manageability                    : OFF
> ME Capability: IntelR Anti-Theft (AT)                     : OFF
> ME Capability: IntelR Capability Licensing Service (CLS)  : ON
> ME Capability: IntelR Power Sharing Technology (MPC)      : OFF
> ME Capability: ICC Over Clocking                          : ON

I think ICC means "Integrated Clock Controller" - I'm not sure what Over Clocking it would do.

> ME Capability: Protected Audio Video Path (PAVP)          : ON

Used for Microsoft PlayReady DRM, probably also HDCP and allows ME to intercept all graphic rendering, I think.

> ME Capability: IPV6                                       : ON
> ME Capability: KVM Remote Control (KVM)                   : ON

OOB Remote KVM capability

> ME Capability: Outbreak Containment Heuristic (OCH)       : OFF
> ME Capability: Virtual LAN (VLAN)                         : ON
> ME Capability: TLS                                        : ON
> ME Capability: Wireless LAN (WLAN)                        : ON

czeej, 2020-03-09:
Would anyone know if an associated wifi card could pass on Wireless credentials to intel ME? 
Also I am curious if intel ME Minix Kernel has keylogging capabilities which would defeat most disk encryption.

flawedworld, 2020-03-10:
The odds of the ME being equipped with a backdoor is slim to none. The odds of there being an exploit based around the ME's priviledges, somewhat high, as history has shown. The ME highly likely does not contain the capabilities you mention. All the ME is, is potential attack surface and this project merely reduces it down to as much as possible. Just because it is a black box does not mean it instantly is malicious. The key issue of the ME is that it is poorly implemented.

czeej, 2020-03-12:
While I believe that for sure. I've seen intel staff that seemed excited about AMT when it dropped. However, It has such low level access. I am always skeptical of trusting it. 

Especially in laptop space becuase much of hardware is hard to change so firmware vendors know what to develop for. Firmware can be particularly nasty. Slightly off the ME topic but check it out. Vulnerable possibly several implementations and versions.

[https://hacked.com/israeli-researchers-turn-speakersheadphones-eavesdropping-microphones/](url)

flawedworld, 2020-03-18:
For now the general consensus is that you use pre skylake hardware to avoid the majority of intel exclusive vulnerabilities (Intel are essentially recycling skylake constantly since it's release) and AMD Zen 2 has the least amount of attack surface at the hardware level implementation. Sure it's got the PSP, but the PSP does not have network access, which is the key attack surface in the Intel ME (IMHO). The PSP is also based on TrustZone a more proven TEE system (IMHO) used in most if not nearly all modern android based devices. With all signed device firmware you are placing trust in the distributor/manufacturer. It's a matter of user control and unless you are actively being targeted specifically it is unlikely that said trust will be used to impact you specifically.

ginto37, 2020-03-19:
Gabe Gałązka:
> For now the general consensus is that you use pre skylake hardware to avoid the majority of intel exclusive vulnerabilities (Intel are essentially recycling skylake constantly since it's release) [...]
Guys, OP here. This is not on-topic for this Issue. Please move this
discussion to a new Issue.
Thanks.

skochinsky, 2020-03-19:
To answer the question as asked, "capabilities" in this case refers to "available/enabled features" in the ME firmware. The stock ME firmware from Intel comes with many modules the set of which changes based on the edition (Consumer/Corporate/Slim) and they can further be disabled by the OEM or at provisioning time with various granularity. You can find toggles for most of them in the OEM-targeted Intel Flash Image Tool (FIT).

ginto37, 2020-03-21:
Igor Skochinsky:
> To answer the question as asked, "capabilities" in this case refers to "available/enabled features" in the ME firmware. The stock ME firmware from Intel comes with many modules the set of which [...]
That's interesting, but I was hoping to find out more about what each
module actually does. You sound knowledgeable on the topic, I don't
suppose you could fill in some of the gaps? Maybe you have a link to
official info about each module?