corna/me_cleaner #296: Update Wiki to reflect that all Mr. ChromeBox Coreboot Firmewares should work with cleaned ME

moriel5, 2019-09-16:
Having cleaned the TXT (I know people refer to it as ME, however Atoms and Celerons, and some Pentiums, utilize Intel's TXT, which is sort of a cut-down version of the ME) on my sister's Acer ChromeBook CB11-131, which runs Mr. ChromeBox's CoreBoot firmwares, and having read that Mr. ChromeBox himself attests that all his ChromeBooks running his firmware work fine after being cleaned with ME Cleaner.

moriel5, 2019-10-06:
Thanks, that should be noted.

Though I believe that KabyLake-Y utilizes the ME, rather than the TXT, despite being based on the (refined) Atom architecture.
In general, KabyLake is newer, so perhaps the bugs are due to some of the newer quirks.

myfreeweb, 2019-10-06:
Well, "all Mr. ChromeBox Coreboot Firmewares" does not suggest only caring about the Atom-based ones :)

> despite being based on the (refined) Atom architecture

Wikichip suggests it has real Kaby Lake cores.

moriel5, 2019-10-06:
> Well, "all Mr. ChromeBox Coreboot Firmewares" does not suggest only caring about the Atom-based ones :)

Sorry, I don't even remember whether I was too vague or whether I had misunderstood Mr.ChromeBox's comment on the matter.
I'll be home in a few days, so I'll reread his comment on the matter (and if need be, I'll ask them myself), and update this thread.

> Wikichip suggests it has real Kaby Lake cores.

It does, however it is the low power Kaby Lake cores, i.e. Atom-Z>Core-M>Core-Y.
On the one hand, it has some of the performance boosts and security and other features  from the Core series, while on the other hand, it is based on the Atoms' power effiency.

HardenedVariety, 2024-08-08:
@moriel5

Is it possible to run me_cleaner on coreboot image and flash internally on chromebooks?