corna/me_cleaner #299: im use t430 testing not working?

byaktakenzie, 2019-10-04:
lcd get very dark/don't have light and keyboard not working after flashing... im restore stay blank( laptop get power from keyboard but lcd very dark )

byaktakenzie, 2019-10-04:
can i get solution? i flash again my backup but don't fix

macromorgan, 2020-01-23:
If the battery was still present or power applied while you reconnected the LCD cable after reassembly of the laptop on a T430 you fried a fuse on the motherboard.  The fuse is F7 and while you technically can solder a jumper in place to bypass the fuse you might also consider getting a new motherboard.

Can't speak to the keyboard other than maybe the ribbon cable isn't reconnected properly?