corna/me_cleaner #304: Does ME >= 12 ignore the HAP bit?

Matthew-Bradley, 2019-10-24:
Was reading https://docs.google.com/document/d/1NRXqXcLBp5pFkHiJbrLdv3Spqh1Hu086HYkKrgKjeDQ/edit#

They note that on ME version >= 12, it ignores the HAP bit.

dt-zero, 2019-10-25:
This is not true. I have successfully disabled Intel ME 12 using the High Assurance Platform bit. It's position changed on the latest platforms compared to where it used to be located inside the PCH strap region. See my PR https://github.com/corna/me_cleaner/pull/282

Espionage724, 2019-11-15:
> This is not true. I have successfully disabled Intel ME 12 using the High Assurance Platform bit. It's position changed on the latest platforms compared to where it used to be located inside the PCH strap region. See my PR #282

Thanks a lot for this PR! I have a Acer Predator Helios 300 laptop with Coffee Lake and ME 12. I used the PR `.py`, and successfully disabled the HAP bit on my dumped BIOS.

Laptop boots fine, and I confirmed ME disabled from a few sources (BIOS no longer presents a TPM option, no HECI PCI device present in `lspci`, and `intelmetool` can't find ME PCI device anymore)