corna/me_cleaner #305: Not working on Samsung NP300E5C-A01PL (Intel® Core™ i5-3210M)

NP-300, 2019-11-04:
I changed processor Intel® Pentium® Processor B960 to Intel® Core™ i5-3210M in Samsung NP300E5C-A01PL. After change laptop power off every 30 minutes even after apply Me_Cleaner witch parameter -S, witch no arguments and witch parameter -s. I have chip Macronix MX25L6406E and I don’t have problems witch flashing it. Unfortunately in this laptop I have Chipset HM70 which is blocked to work longer than 30 minutes witch processor better than Pentium. It is possible to deal witch this issue using Me_Cleaner?

yhaenggi, 2019-11-04:
how did you flash it?

NP-300, 2019-11-04:
I used Raspberry Pi Model 3B+ and this instruction: https://github.com/corna/me_cleaner/wiki/External-flashing,
Option: “Neutralize Intel ME” (without coreboot)

Stitch626, 2020-08-07:
Which cpus did you test? Cuz you changed from Sandybridge to Ivybridge <- That may be your problem. As far as I know you need a minimum ME version to get support for Ivybridge so thats probably the issue. Do you already use the newest bios? Is there official support for those cpus?

NP-300, 2020-08-08:
Never mind. I change motherboard to different model witch better chipset that supports better processors and now everything work properly. This laptop was sold witch different processors but with Pentium this laptop had different motherboard with different chipset then with other processors.

Stitch626, 2020-08-08:
Okay, so it's no longer an issue and you should close this one :)

closed 2020-08-08 by NP-300