corna/me_cleaner #311: me cleaner brakes wired internet on haswell

maxbla, 2020-01-14:
CPU: Haswell (Intel i5 4670k)
Motherboard: ASRock Z97M-ITX/AC
Ethernet controller: Qualcomm Atheros QCA8171 Gigabit Ethernet (rev 10) (from `lspci` output)

After running me_cleaner, the Ethernet interface doesn't show up to multiple Linux OSs. To be clear, running `ip link` or `ifconfig`, the enp2s0 device simply isn't listed (and doesn't work). This is true on all of Ubuntu, Arch and Fedora. Strangely, WiFi does work. Reflashing the original firmware immediately fixed the issue.

After some experiments, this ethernet problem occurs when using the `-S` flag and when using no flags, but when using the `-s` flag, ethernet works.

I used an external CH341A flasher.

65a, 2020-01-29:
Same issue here on Kabylake-U i5 device. I have tried a variety of options in Flash Image Tool as well. My symptoms are the same as yours. In OEM bios with -s, it works. In OEM bios with -S, first ethernet device is missing. In Coreboot, the same is true as OEM bios. Notably, Coreboot logs seem to indicate the entire PCI Express root port #1 goes missing.

czeej, 2020-03-22:
Have you installed a Ethernet card on a pci slot? Will that work for you? I always thought that is a good option for desktop instead of using the provided NIC.