corna/me_cleaner #312: ThinkPad T480

CodeDragon5793, 2020-02-06:
Okay so something unexpected happened. Upon second reboot of the system, the computer emits a loud beep, complains about bottom cover tamper detection, and then flashes a CMOS error very quickly before shutting down. Repowering up the device and quickly entering the BIOS allows me to reset the time, and successfully boot into the OS, but the HAP bit is not longer set to 1.

CodeDragon5793, 2020-02-07:
So I have tried all of the possible commands specified in this documentation. I tried -S, -s, and no arguments. They all seemed to work as per the output on my Raspberry Pi's terminal. Then I would boot, and it would take a little longer to boot that time, but when I would use the coreboot intelme tool all three times the ME had failed to be removed. I have a theory that the EC is resetting the Intel ME, or am I doing something wrong?

closed 2020-02-07 by CodeDragon5793

skochinsky, 2020-02-08:
some BIOSes have a copy of the ME firmware in the BIOS region and flash it if the ME firmware does not respond properly (which is the case after me_cleaner). If this is your case, you will need to find that copy of the firmware in the BIOS region and replace it with the cleaned image, though it's possible this may lead to a boot loop (so it may be better to remove the UEFI module which performs this reflashing).