corna/me_cleaner #314: ASUS Z97-A - Mostly working

flawedworld, 2020-02-14:
Running me_cleaner with -S (Removing ME modules where possible and HAP bit) leaves the board fully functional, however the TPM functionality is lost, and is stuck in 'SUPPORT TURNED OFF' in the BIOS.

Also, removing the CMOS battery causes a reboot loop which is only fixable by a reflash? Bit odd.

Tested on latest beta BIOS 3503 (Which is stable bios + spectre/meltdown patches)

czeej, 2020-03-08:
I think you might want to add -d as well. 
Your I had similar issues where I could remove network stack from ME. But I believe Dell bios has its' own.

ValoWaking, 2020-04-27:
See on your FSB speed, is it 99.75 mhz?

flawedworld, 2020-05-02:
> See on your FSB speed, is it 99.75 mhz?

Sorry for the lateness, but this board has issues with correctly setting an accurate BLCK (This generation of CPU's does not have a FSB technically).

I believe this is down to the quartz clock generator (I think thats the right term) not hitting 100mhz perfectly. Either way its nothing to worry about. It is nothing to do with ME.

ValoWaking, 2020-05-02:
Do u can create your cpu-z screenshot?

ValoWaking, 2020-05-02:
BLCK is down to 99.75 because ICC profile is deleted by me_cleaner...

closed 2021-09-25 by flawedworld