corna/me_cleaner #315: OutOfRegionException when cleaning a shrunk descriptor+me image

p4block, 2020-02-19:
I've been running coreboot on my x230 for 2 years now updating once in a while. Decided I wanted more space, so I followed the "internal flashing with coreboot" wiki page to generate a modified descriptor and fully shrunk me.bin. Rebuilt coreboot and flashed, but noticed something weird:

The full firmware now crashes me_cleaner.py
```
me_cleaner.py coreboot.rom                                                                                                                                       
Full image detected
The ME/TXE region goes from 0x3000 to 0x1b000
Found FPT header at 0x3010
Found 1 partition(s)
Found FTPR header: FTPR partition spans from 0xd00 to 0xcad00
ME/TXE firmware version 8.1.30.1350
Public key match: Intel ME, firmware versions 7.x.x.x, 8.x.x.x
The AltMeDisable bit is SET
Reading partitions list...
 FTPR (      no data here      , 0x000ca000 total bytes): nothing to remove
Removing partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0x7c)...
Reading FTPR modules list...
 UPDATE           (LZMA   , 0x04d1f2 - 0x04d3b0       ): Traceback (most recent call last):
  File "/usr/local/bin/me_cleaner.py", line 799, in <module>
    check_and_remove_modules(mef, me_end, ftpr_offset,
  File "/usr/local/bin/me_cleaner.py", line 367, in check_and_remove_modules
    end_addr = remove_modules(f, mod_headers, offset, me_end)
  File "/usr/local/bin/me_cleaner.py", line 176, in remove_modules
    f.fill_range(offset, end, b"\xff")
  File "/usr/local/bin/me_cleaner.py", line 95, in fill_range
    raise OutOfRegionException()
__main__.OutOfRegionException
```
_I flashed anyway and the laptop now blackscreens on boot (but doesn't do the usual coreboot panic with flashing leds, so it could be an unrelated breakage) I have no way to confirm until tomorrow when I can flash externally._

p4block, 2020-02-21:
Due to lack of time I just flashed a known working backup. It's unknown if the computer not booting was caused by whatever is also causing this bug, or something else entirely. I would guess it's related to this because little has changed in the coreboot repo since my last build ~a month ago.

czeej, 2020-03-25:
You need at least a bup module in FTPR.. It said no data here in your FTPR.
If you watch the conference they did on ME. BUP is the boot portion of the firmware. 
Odd how your image has an empty ftpr