corna/me_cleaner #318: Opportunities with CSME vulnerability?

lenzj, 2020-03-05:
I saw the article below when reading news today.  I can't tell whether this is an opportunity for me_cleaner / coreboot to leverage or whether it's the nail in the coffin for any security conscious user wanting to use an Intel computer affected by this.  The key statement in the article that caught my eye was the following: "In ROM, this vulnerability also allows for arbitrary code execution at the zero level of privilege of Intel CSME.  No firmware updates can fix the vulnerability."

https://www.ptsecurity.com/ww-en/about/news/unfixable-vulnerability-in-intel-chipsets-threatens-users-and-content-rightsholders/

ngortheone, 2020-03-05:
I was going to ask the same question. +1
https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2019-0090

emanruse, 2020-03-10:
I have a system on which I have applied me_cleaner successfully some years ago.
Am I still affected by this vulnerability?

IOW: Is a system with disabled ME affected by this at all? If yes/maybe: How does one check? (no, I don't have a ME menu in my BIOS, never had one)

ngortheone, 2020-03-11:
@emanruse yes, the vulnerability is in hardware and it allows extracting root cryptographic key (Chipset Key) given full physical access to the device (like open the case and touch the mothreboard) 

Every Intel chipset in last 5 years is affected.

This vulnerability presents an interest in context of ME_cleaner because given access to chipset key it is possible to wipe ME completely (the same way it is done on old devices like Thinkpad X200) and replace UEFI bios with coreboot/anything else - you can sign any payload and system with trust it.

Or so at least people in this thread hope. We are waiting on experts to weigh in.

https://securityaffairs.co/wordpress/99120/hacking/intel-cve-2019-0090-flaw.html

emanruse, 2020-03-11:
Thanks for explaining.

So this is a vulnerability which is related only to physical access to the hardware, i.e. cannot be exploited remotely (through a network)?

> Every Intel chipset in last 5 years is affected.

The motherboard on the machine I use here was purchased in 2012. In the user guide I read it has Intel Z77 Express Chipset (LGA1155). So it never really had "CSME", it has just ME (which was disabled through me_cleaner as I mentioned).

So - am I affected?

> This vulnerability presents an interest in context of ME_cleaner because [...]

Interesting.

As for the link you provided, I read there:

"Only Intel 10th generation processors, Ice Point chipsets and SoCs, are not affected by the flaw."

If one reads that literally one could think that even old systems (manufactured before ME even existed) are affected.

z3r0p0int3r, 2020-04-04:
As I understood, the main point of this subject was to bring attention on yet another method of deblobbing IME without using external programmers and soldering. It is important to have such methods because it's always a big headache relying solely on external programming skills, because not everyone has such skills/time/opportunity to do this. Personally I prefer avoiding external programmers when using `me_cleaner` on laptops when it's possible to dump and flash the whole BIOS using the internal programmer via `fpt`.

For now generally known methods to dump and flash IME without using external programmers are:

- HDA_SDO (Flash Descriptor Security Override) pins shorting on Intel HDA chips to unlock the FD (works on almost all laptops that have an intel high definition audio chip)

- Motherboard switches (mostly on desktops, laptops don't have this)

- Hidden BIOS-UEFI option to unlock ME like [here](http://forum.notebookreview.com/threads/guide-dell-precision-m6800-m4800-sbios-mod.788481/)

All those methods rely on using Intel Flash Programming Tool (`fpt`) for dumping and flashing the internal chip(s).

This vulnerability opens an opportunity to fulfill this list with yet another method.

lenzj, 2020-04-04:
@z3r0p0int3r yes that could be one benefit, however the "holy grail" in my opinion would be for the hardware owner to obtain the root cryptographic key for the ME to enable complete wipe or alternately reprogramming the ME with open source code such as coreboot etc.

The current state with me_cleaner is that it erases a large chunk of the ME (which is great), however there is still a small core that is protected by a cryptographic key which cannot currently be wiped.  It's unknown what this residual encrypted ME code does, or whether there are further vulnerabilities in it that unscrupulous individuals or agencies could still exploit against owners of said hardware.