corna/me_cleaner #319: Is any of these symptoms related to a neutralized Intel ME? [Ivy Bridge, HM77]

mbc07, 2020-03-06:
So, I own a HP ENVY dv6-7300 laptop, which has the Intel HM77 chipset and, from what I could gather poking around a BIOS update file for the same BIOS version I'm currently running, Intel ME Firmware 8.1.0.1248. 

The motherboard from this laptop failed recently and I got it swapped with a pre-owned one, that's when I started to notice some strange behaviors that never happened on the previous (now dead) board:

* During a cold boot (laptop completely turned off but plugged to AC power and/or battery installed), when I press the power button, the laptop will turn on but the the screen remains off. Only after a delay of 10 - 15 seconds is when the screen will turn on and Windows will proceed booting shortly after. 
* When that screen off delay during power up happens, Windows boot time is significantly longer, taking about 2 minutes compared to the usual 15 - 20 seconds. Then, if you go to the Device Manager after Windows finished booting, Intel Management Engine Interface will have an yellow exclamation mark with error code 10 and the message STATUS_DEVICE_POWER_FAILURE. 
* Digging on Windows Event Viewer, I can also see "Intel(R) Management Engine Interface driver has failed to perform handshake with the Firmware (FWSTS0: 0x00304181, FWSTS1: 0x153B0100)" followed by "The Intel(R) Management Engine Interface is being disabled".
* After that, if you reboot the laptop (or if you shut it down but turn it on again within the next 5 to 10 minutes), there won't be any delay during the power up nor the Windows boot will take longer, but also, "Intel Management Engine Interface" will be completely absent from the Device Manager, instead showing with a code 45 (device isn't currently attached to the system).

I also tried to run Intel's own MEInfo tool to shed some light on the actual status of Intel ME on this system, but running both the Windows and also the EFI version from the BIOS UEFI shell resulted always in a generic "failed to communicate to MEI" (or something along those lines, don't remember the exact wording now).

Given this motherboard was pre-owned, I suspected the previous owner might have used me_cleaner. So, my question is, how exactly a computer which got Intel ME purged with me_cleaner behaves? Are any of those symptoms similar to a system with Intel ME neutralized? And finally, does not having a working Intel ME cause any negative impact on other system components/features?

mbc07, 2020-03-07:
So, after some digging on Win-Raid forum, turns out it was just a corrupted ME firmware. I was able to fix it by temporarily disabling SPI flash lock (on this particular model you can do this by pressing Win + Left Arrow + Right Arrow + Power button when the laptop is turned off) then modifying the INI file of the HP BIOS update (SP77453) to also flash ME region (add `ME=1` under `[Region]` section of `config2.ini`).

The relevant piece of info for me_cleaner certainly is the Win + Left + Right + Power key combo to temporarily disable the SPI flash lock. Since it allows fully reading/writing to all parts of the firmware (including Intel ME region) while the lock is disabled, this basically enables Internal Flashing of cleaned images on all HP Envy dv6-7000 and dv7-7000 laptop series, for anyone interested...

closed 2020-03-07 by mbc07

mbc07, 2023-06-06:
The BIOS update provided by HP includes the ME region, but it's skipped during flashing unless you extract the update and edit the config INI file as I mentioned in the previous comment. 

Also don't forget to turn the laptop on with the hotkey combination, otherwise the ME region will be locked and flashing will fail...

dstroq, 2023-06-07:
thank you very much, on the third attempt it turned out to update the ME region.