corna/me_cleaner #321: Dell Latitude E7450 potential bios EC change, unable to reflash safely.

czeej, 2020-03-08:
Every dump I get now tells me ME is corrupted or missing, they are not identical between two dumps as well. I imagine this implementation is quite nasty or maybe not.. The wifi card is hard to remove without full disassembly.

note lspci does not register any ME region when booted in. 

However, now I can't seem to get a valid dump. It is telling me specific memory regions are bad. 

I never used descriptor modifier on ME cleaner either which I believe I should have done.

Sticky keys seems to like to activate for no apparent reason as well. I imagine something is going on even with ME kernel invisible to the OS. It was not removed as it wont boot. Only partition I could get was ME network stack but I believe dell has it's own and without the descriptor region modified I'm afraid it's no good. 

Can't seem to get valid dumps anymore. Not sure why this is, anyone more experienced able to help me trouble shoot?

mostav02, 2020-03-22:
I have never tried dumping this laptop in-circuit, but had a friend unsoldering and dumping 2 chips off-circuit for me via an external programmer called BeeProg+  (we dumped 4MB and 8MB chips)

The dumps were correct but we made them before [August 2017](https://github.com/corna/me_cleaner/commit/93e65775bb2ce4ac5df3ae29b8c14f189529bcfd), so there was no `-s` flag in me_cleaner and HAP&AleMeDisable undiscovered, and since `vPro`-enabled laptops (including E7450) have Boot Guard enabled, using old versions of me_cleaner before August 2017 would always fail, which what exactly happened back on the date during our attempt - the laptop was turning on and turning off instantly (typical BG-enabled BIOS behavior).

I would have liked to try flashing it again with `-s` which would only modify the FD without touching ME region, but unfortunately I have no external programmer to do so right now, nor my friend is reachable anymore.

However I will drop these dumps here in case they may help to debug yours and other problems. The 8MB one is a correct one, I don't know what 4MB chip is for, but it was a nearest chip so we dumped it too, just in case. The BIOS on this one is A13.

[Dell_E7450.A13.spi_dumps.zip](https://github.com/corna/me_cleaner/files/4365986/Dell_E7450.A13.spi_dumps.zip)

I hope it will help somehow.

mostav02, 2020-03-24:
I have never tried dumping this laptop in-circuit, but had a friend unsoldering and dumping 2 chips off-circuit for me via an external programmer called BeeProg+  (we dumped 4MB and 8MB chips)

The dumps were correct but we made them before [August 2017](https://github.com/corna/me_cleaner/commit/93e65775bb2ce4ac5df3ae29b8c14f189529bcfd), so there was no `-s` flag in me_cleaner and [HAP&AltMeDisable](https://github.com/corna/me_cleaner/commit/ced3b46ba2ccd74602b892f9594763ef34671652) undiscovered, and since `vPro`-enabled laptops (including E7450) have Boot Guard enabled, using old versions of me_cleaner before August 2017 would always fail, which what exactly happened back on the date during our attempt - the laptop was turning on and turning off instantly (typical BG-enabled BIOS behavior).

I would have liked to try flashing it again with `-s` which would only modify the FD without touching ME region, but unfortunately I have no external programmer to do so right now, nor my friend is reachable anymore.

However I will drop these dumps here in case they may help to debug yours and other problems. The 8MB one is a correct one, I don't know what 4MB chip is for, but it was a nearest chip so we dumped it too, just in case. The BIOS on this one is A13.

[Dell_E7450.A13.spi_dumps.zip](https://github.com/corna/me_cleaner/files/4365986/Dell_E7450.A13.spi_dumps.zip)

Hope it may help in some way. Maybe you can compare your dumps against those and make some conclusions.

czeej, 2020-03-25:
If you try again I recommend removing the Descriptor region. As attempt to prevent the Dell bios from interacting with the ME kernel. I am only speculating and no nothing of what is going on under the hood. However, for me to be able to remove the network partition and still have successful boot and the presence of Root certificates in the Dell bios tells me the Dell bios has it's own network stack. Making the Intel ME network stack redundant anywho. 

Again, thanks for the dump I could provide my successful dumps however, Now I cannot get two matching dumps. The External programmer/ Just Raspi SPI to bread board is not working on this laptop. Some kind of block protection or real corruption in the eeprom. Again, external programmer is giving me over current warnings. Potential corruption or hardware issues. 
All dumps were in circuit but as you know you must remove battery to get under the hood anyways. I usually like to hit the power button as well a few times without battery to discharge extra electrons. However, I may of skipped that step a few times.

I feel de-soldering and putting a new pre flashed winbond on would be a potential solution. However, as far as I'm concerned the laptop is not comfortable to use for me at the current moment. Thanks for the comment. Potentially virtualization features in bios are enabling some boot guard but as it turns out it is actually disabled atm. I'm not sure what would be causing specific memory addresses to be consistently giving me warnings on dumps. Usually, pretty early in the memory 0x (I can't remember exact address, maybe 0x32) not sure anymore. I've since sold the laptop to a windows user. I figured the 4Mb would be Dell bios but I could be wrong. Also maybe be Computrace, if you also have the same firmware from Absolute software.

mostav02, 2020-04-09:
> Potentially virtualization features in bios are enabling some boot guard but as it turns out it is actually disabled atm.

Generally all of those models are BG enabled and have both _Measured Boot_ and _Verified Boot_ enabled, such as on mine. I got 2 of those laptops bought in different countries and both of them are BG MB&VB enabled. It's actually very weird that you had BG disabled on E7450 and maybe it was a possible source of your suspicious problems.

Today I've got an external programmer and reconfirmed that this laptop **won't power on** once you modify the BIOS or ME region. The only possible way of deactivating ME here is by enabling the AltMeDisable bit, which actually works pretty well as I've just did and tested it.

Anyway, what's good is I've figured out that **it's possible to disable ME on E7450 without using an external programmer!** I know it would be not interesting to you since you got rid of the laptop, but I will just leave it here still so anyone wouldn't need to incur the same problems as we had.

What is needed: a screwdriver, a USB mouse&keyboard and this laptop running Windows 7.

- Download Intel ME System Tools v10.0 ( e.g. [here](https://mega.nz/#!zA8hQBZT!IuRKy4ra2dBKsLHDR-bBhsWwYpSvG65TwfOQj8SCqKA) )
- Remove the keyboard with the top cover and locate the [Realtek ALC3235](https://imgur.com/FVMBdYR) chip which is the _Intel High Definition Audio_ chip.

[image]
([backup image link](https://i.imgur.com/pteONWM.jpg))

- Short the pins 1 and 5 on the ALC3235 with something like [this](https://imgur.com/m0DQZye), then power on the laptop while keeping pins shorted (the power button is accessible without the top panel)
- If done correctly, you will be able to dump and flash the full firmware using _Flash Programming Tool_ from _Intel ME System Tools v10.0_

In this case we only need to dump, modify and flash the Flash Descriptor.

- Dumping the FD: `fptw64.exe -DESC -D dumpedFD.bin`
- Enabling the AltMeDisable in FD: can be modified using the `ifdtool` (the `-M` flag) or `me_cleaner` with the `-s` flag.
- Flashing the modified FD: `fptw64.exe -DESC -F dumpedAndModifiedFD.bin`

The output may look like this upon flashing:

> C:\fucknsa\Intel ME System Tools v10.0 r7\Flash Programming Tool\WIN64>fptw64.exe -DESC -F dumpedAndModifiedFD.bin
> Intel (R) Flash Programming Tool. Version:  10.0.30.1054
> Copyright (c) 2007 - 2014, Intel Corporation. All rights reserved.
>
> Platform: Intel(R) Premium Express Chipset
> Reading HSFSTS register... Flash Descriptor: Valid
>     --- Flash Devices Found ---
>     W25Q64BV    ID:0xEF4017    Size: 8192KB (65536Kb)
>     W25Q32BV    ID:0xEF4016    Size: 4096KB (32768Kb)
> 
> PDR Region does not exist.
> 
> - Reading Flash [0x001000]    4KB of    4KB - 100% complete.
> - Erasing Flash Block [0x001000] - 100% complete.
> - Programming Flash [0x001000]    4KB of    4KB - 100% complete.
> - Verifying Flash [0x001000]    4KB of    4KB - 100% complete.
> RESULT: The data is identical.
> 
> FPT Operation Passed
> 

After rebooting you will not see the ME interface anymore anywhere. 

- **[!]** You can safely upgrade or downgrade the BIOS and Dell BIOS updates will follow the AltMeDisable flag and won't bother you with ME Firmware updates anymore: they will only upgrade BIOS and skip the usual ME firmware updates (amazing).

_offtopic: I've noticed that the BIOS in E7450 has a lot of cool hidden options, would be good to know how to enable them - [E7450 BIOS SetupPrep UEFI IFR extracted.txt](https://github.com/corna/me_cleaner/files/4454359/SetupPrep.IFR.txt)_

mostav02, 2020-04-09:
> Potentially virtualization features in bios are enabling some boot guard but as it turns out it is actually disabled atm.

Generally all of those models are BG enabled and have both _Measured Boot_ and _Verified Boot_ enabled, such as on mine. I got 2 of those laptops bought in different countries and both of them are BG MB&VB enabled. It's actually very weird that you had BG disabled on E7450 and maybe it was a possible source of your suspicious problems.

Today I've got an external programmer and reconfirmed that this laptop **won't power on** once you modify the BIOS or ME region. The only possible way of deactivating ME here is by enabling the AltMeDisable bit, which actually works pretty fine as I've just did and tested it.

Anyway, what's good is I've figured out that **it's possible to disable ME on E7450 without using an external programmer!** I know it would be not interesting to you since you got rid of the laptop, but I will just leave it here still so anyone wouldn't need to incur the same problems as we had.

What is needed: a screwdriver, a USB mouse&keyboard and this laptop running Windows 7 (unfortunately there is no Intel's FPT for Linux).

- Download Intel ME System Tools v10.0 ( e.g. [here](https://mega.nz/#!zA8hQBZT!IuRKy4ra2dBKsLHDR-bBhsWwYpSvG65TwfOQj8SCqKA) )
- Remove the keyboard with the top cover and locate the [Realtek ALC3235](https://imgur.com/FVMBdYR) chip which is the _Intel High Definition Audio_ chip.

[image]
([backup image link](https://i.imgur.com/pteONWM.jpg))

- Short the pins 1 and 5 on the ALC3235 with something like [this](https://imgur.com/m0DQZye), then power on the laptop while keeping pins shorted (the power button is accessible without the top panel)
- If done correctly, you will be able to dump and flash the full firmware (All 12MB region on both chips) using _Flash Programming Tool_ from _Intel ME System Tools v10.0_

In this case we only need to dump, modify and flash the Flash Descriptor.

- Dumping the FD: `fptw64.exe -DESC -D dumpedFD.bin`
- Enabling the AltMeDisable in FD: can be done using the `ifdtool` (the `-M` flag) or `me_cleaner` with the `-s` flag.
- Flashing the modified FD: `fptw64.exe -DESC -F dumpedAndModifiedFD.bin`

The output may look like this upon flashing:

> C:\fucknsa\Intel ME System Tools v10.0 r7\Flash Programming Tool\WIN64>fptw64.exe -DESC -F dumpedAndModifiedFD.bin
> Intel (R) Flash Programming Tool. Version:  10.0.30.1054
> Copyright (c) 2007 - 2014, [...]

After rebooting you will not see the ME interface anymore anywhere. 

- **[!]** You can safely upgrade or downgrade the BIOS and Dell BIOS updates will follow the AltMeDisable flag and won't bother you with ME Firmware updates anymore: they will only upgrade BIOS and skip the usual ME firmware updates (amazing).

_offtopic: I've noticed that the BIOS in E7450 has a lot of cool hidden options, would be good to know how to enable them - [E7450 BIOS SetupPrep UEFI IFR extracted.txt](https://github.com/corna/me_cleaner/files/4454359/SetupPrep.IFR.txt)_

czeej, 2020-04-09:
Also to be noted for other I may of run into Signing issues either because I removed the NFTP partition. Which all partitions are signed so maybe it messed up the signature by removing the stack. It boots without it. I did soft disable but wanted to remove as much as I could.
 OR I had a short circuit and. I had my raspi hard reset when connecting the clip. This was the only board this had happened on. However, it was reading fine after that it was only with continued use that when I wanted to re-dump I couldn't 

Anyway. that is a cool discovery. wonder how you found out that shorting the Real Tek smd. Would allow you to get a dump.. Odd you'd think the two wouldn't be related ;)

mostav02, 2020-04-09:
> wonder how you found out that shorting the Real Tek smd. Would allow you to get a dump.. 

There: https://www.win-raid.com/t3553f39-Guide-Unlock-Intel-Flash-Descriptor-Read-Write-Access-Permissions-for-SPI-Servicing.html

This unlocks all the ROM regions.

By the way, I've dumped the full 12MB ROM from 2 chips (W25Q64 as 8MB + W25Q32 as 4MB) and noticed that they work as a stripped storage (kind of _raid0_). I've made some binary comparison between the dumps I had made with ext. programmer and those I made now via FPT. 
Basically when you get the dump of one of them you get the incomplete info, that's why I was unsure what was the purpose of 4MB one before I discovered FPT these days. 

The best way of flashing and reading them is internally because it keeps the data consistent. However, in this case we only need FD which is first 4kb of W25Q64 (8MB).

mostav02, 2020-04-09:
> wonder how you found out that shorting the Real Tek smd. Would allow you to get a dump.. 

There: https://www.win-raid.com/t3553f39-Guide-Unlock-Intel-Flash-Descriptor-Read-Write-Access-Permissions-for-SPI-Servicing.html (search for this string: _E1. HDA_SDO/GPIO33_)

This unlocks all the ROM regions.

By the way, I've dumped the full 12MB ROM from 2 chips (W25Q64 as 8MB + W25Q32 as 4MB) and noticed that they work as a stripped storage (kind of _raid0_). I've made some binary comparison between the dumps I had made with ext. programmer and those I made now via FPT. 
Basically when you get the dump of one of them you get the incomplete info, that's why I was unsure what was the purpose of 4MB one before I discovered FPT these days. 

The best way of flashing and reading them is internally because it keeps the data consistent. However, in this case we only need FD which is first 4kb of W25Q64 (8MB).

mostav02, 2020-04-09:
> wonder how you found out that shorting the Real Tek smd. Would allow you to get a dump.. 

There: http://forum.notebookreview.com/threads/guide-dell-precision-m6800-m4800-sbios-mod.788481/#post-10204348 
It's mentioned as "HDA_SDO" jumper or "Flash Descriptor Security Override" in some original sources such as:

https://www.intel.co.uk/content/dam/doc/datasheet/io-controller-hub-10-family-datasheet.pdf (page 78)
https://www.intel.co.uk/content/dam/www/public/us/en/documents/datasheets/6-chipset-c200-chipset-datasheet.pdf (page 91)
https://github.com/ptresearch/me-disablement/blob/master/How%20to%20become%20the%20sole%20owner%20of%20your%20PC.pdf (slide 17)
http://www.corus.pro/pilotes/CorusX/X37/XP/ME/SPI%20Programming%20Guide.pdf
https://libreboot.org/docs/hardware/gm45_remove_me.html

This seems to be an universal method of enabling a security override on most Intel platform that use the High Definition Audio codec.
It unlocks all the ROM regions.

By the way, I've dumped the full 12MB ROM from 2 chips (W25Q64 as 8MB + W25Q32 as 4MB) and noticed that they work as a stripped storage (kind of _raid0_). I've made some binary comparison between the dumps I had made with ext. programmer and those I made now via FPT. 
Basically when you get the dump of one of them you get the incomplete info, that's why I was unsure what was the purpose of 4MB one before I discovered FPT these days. 

The best way of flashing and reading them is internally because it keeps the data consistent. However, in this case we only need FD which is first 4kb of W25Q64 (8MB).

czeej, 2020-04-09:
I'm also curious what the 32 bit flash was in 4Mb storage. That one definitely had read protections/ block protection even while using an external programmer. I assumed it was part of the dell bios potentially where computrace is stored