corna/me_cleaner #323: Why a full dump is required with `-s`?

mostav02, 2020-04-10:
I've noticed that `me_cleaner` requires a full dump when used with `-s`.

I did some comparison of the full dump processed with `ifdtool -M 1` and  `me_cleaner -s` and turns out what is being modified in both cases is only a Flash Descriptor. The final hashsums are equal using both tools, thus the algorithm for enabling AltMeDisable is absolutely same.

`ifdtool` doesn't require a full dump and it's possible use it on a flash descriptor binary. It's very useful when flashing via Intel Flash Programming Tool (FPT) instead of an external programmer.

Of course in any scenario when the FD is unlocked it permits making a fulldump and flashing it back, but in case of `Intel Boot Guard`-enabled machines it's a kind of overdo, because we can only do `fptw -DESC -D fd.bin` and flash it back with `fptw -DESC -F fd.bin`, which is only 4kb instead of the big full dump resulting into the same effect as using a full dump.

It would be great if a full dump was optional when used with `-s` on Flash Descriptor binaries.

skochinsky, 2020-04-10:
I think it is necessary to determine the ME version since that affects the strap location IIRC. You can try to either add an extra switch to specify the assumed version or somehow detect it from the descriptor...

mostav02, 2020-04-10:
Yeah I can see in the code that it checks for the `variant` prior to processing the `args.soft_disable_only` conditional code block. 

The determined version is only used in a  `print()` once.

The `-s` code block itself doesn't check for anything related to variant and simply does the bitwise operations such as `ifdtool` does.

Ideally there should be a way to do what you suggested. In case Corna won't invent a better way of doing it, I will submit a PR later.

privacyguy123, 2020-06-25:
I noticed this too - `ifdtool` works on my dumps but `me_cleaner` complains about a corrupted image which it most definitely is not, as I've flashed it to my machine many times successfully.