corna/me_cleaner #326: Clean Nehalem/Westmere ME 6.0

ValoWaking, 2020-04-27:
Hi. I have very important question about Nehalem/Westmere Intel ME 6.0. Can me_cleaner clean all partitions (delete all firmware) on Intel ME 6.0 or not? 

In patch note 1.2v we have:

Wipe the ME6 Ignition firmware images
Fix the removal of the last partition on ME6

Stitch626, 2020-08-07:
You can't "delete all firmware" cuz some regions are required by the chipset itself. The last one with fully removable (in this case replacable ;) ) firmware was the one used on the 775 non-Q chipsets.
So the short answer is NO.

ValoWaking, 2020-08-07:
> 
> 
> You can't "delete all firmware" cuz some regions are required by the chipset itself. The last one with fully removable (in this case replacable ;) ) firmware was the one used on the 775 non-Q chipsets.
> So the short answer is NO.

No it's wrong. Try to download some 1156 bios and wipe it by your self. It's works and fully wipe.

Stitch626, 2020-08-07:
I don't believe that. And even if it does, you may have the 30 minutes problem.

ValoWaking, 2020-08-08:
> 
> 
> I don't believe that. And even if it does, you may have the 30 minutes problem.

WatchDog can be fixed by Flash Image Tool and MMTool