corna/me_cleaner #328: Razer Blade 17 Pro Intel Me

privacyguy123, 2020-05-19:
Keeps spitting out this error for the supplied Intel ME bin file from Razer

`-d, -D, -M, -S and -s require a full dump`

privacyguy123, 2020-06-10:
Maybe someone can take a look at this file?

https://gofile.io/d/Vu5sjL

schattenpanzer, 2020-06-12:
what commands did you type, and what is the full output?

privacyguy123, 2020-06-15:
I typed the commands in the guide & that is the full output!

privacyguy123, 2020-06-15:
Here's -c output:

`ME/TXE image detected
Found FPT header at 0x10
Found 15 partition(s)
Found FTPR header: FTPR partition spans from 0x2000 to 0xe6000
Found FTPR manifest at 0x2250
ME/TXE firmware version 12.0.32.1421 (generation 3)
WARNING Unknown public key 3efc26920b4bee901b624771c742887b
        Assuming Intel ME
        Please report this warning to the project's maintainer!
Checking the FTPR RSA signature... VALID`

schattenpanzer, 2020-06-15:
`python me_cleaner.py -O the_me_cleaned.rom -S your_stock.rom`

This syntax worked for me.

By any chance, are you using Windows? Could you try flashrom and me_cleaner on a GNU/Linux partition?

privacyguy123, 2020-06-16:
Yes this is a Windows laptop ... I'm not sure how a Python script would perform differently on each OS?

schattenpanzer, 2020-06-16:
>Yes this is a Windows laptop ... I'm not sure how a Python script would perform differently on each OS?

They usually don't perform differently, but they could. 

>I noticed my dump (provided by manufacturer) is in .bin format and not .rom? I provided the link to it above.

You're using a BIOS dump from the manufacturer's website, and not from flashrom? My approach would have been to use flashrom, I think you should try that first.

    flashrom -p ch341a_spi -r your_stock.rom

you would need a [usb programmer](https://www.ebay.com/itm/Multifunction-CH341A-Router-USB-Programmer-LCD-Burner-Bios-Board-24-25-Series-AL/292578035653), and since it's a laptop, a [soic8 clip](https://www.ebay.com/itm/SOIC8-SOP8-Flash-Chip-IC-Test-Clips-Socket-Adpter-BIOS-24-25-93-Programmer/223741190827).

privacyguy123, 2020-06-16:
Seeing as my manufacturer supply the official Intel ME firmware dump why would I need to be opening up the laptop and messing about with programmer chips?

skochinsky, 2020-06-16:
You can't use official tools to flash a stripped ME firmware. Also, me_cleaner doesn't work on ME firmware updates but only on flash images.

privacyguy123, 2020-06-16:
> 
> 
> You can't use official tools to flash a stripped ME firmware. Also, me_cleaner doesn't work on ME firmware updates but only on flash images.

With that in mind I'm double confused then - I ran me_cleaner on the manufacturer provided firmware with no switches and got this output 😕 : 

```
Laptop ➜ .\me_cleaner.py .\ME_12.0_Consumer_CNP-H_PCH_Production.bin
ME/TXE image detected
Found FPT header at 0x10
Found 1 partition(s)
Found FTPR header: FTPR partition spans from 0x2000 to 0xe6000
Found FTPR manifest at 0x2250
ME/TXE firmware version 12.0.38.1430 (generation 3)
WARNING Unknown public key 3efc26920b4bee901b624771c742887b
        Assuming Intel ME
        Please report this warning to the project's maintainer!
Reading partitions list...
 FTPR (0x00002000 - 0x0000e6000, 0x000e4000 total bytes): NOT removed
Removing partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0xd8)...
Reading FTPR modules list...
 FTPR.man     (uncompressed, 0x002250 - 0x003000): NOT removed, partition manif.
 rot.key      (LZMA/uncomp., 0x003000 - 0x003440): removed
 kernel       (Huffman     , 0x003440 - 0x012100): NOT removed, essential
 syslib       (Huffman     , 0x012100 - 0x025180): NOT removed, essential
 bup          (Huffman     , 0x025180 - 0x049540): NOT removed, essential
 pm           (Huffman     , 0x049540 - 0x04c2c0): removed
 vfs          (Huffman     , 0x04c2c0 - 0x05b080): removed
 evtdisp      (Huffman     , 0x05b080 - 0x05db00): removed
 loadmgr      (Huffman     , 0x05db00 - 0x062400): removed
 busdrv       (Huffman     , 0x062400 - 0x064ec0): removed
 gpio         (Huffman     , 0x064ec0 - 0x066bc0): removed
 ipc_drv      (Huffman     , 0x066bc0 - 0x0694c0): removed
 prtc         (Huffman     , 0x0694c0 - 0x06a3c0): removed
 policy       (Huffman     , 0x06a3c0 - 0x06fcc0): removed
 crypto       (Huffman     , 0x06fcc0 - 0x09ca40): removed
 heci         (Huffman     , 0x09ca40 - 0x0a2380): removed
 storage      (Huffman     , 0x0a2380 - 0x0ab2c0): removed
 pmdrv        (Huffman     , 0x0ab2c0 - 0x0ad5c0): removed
 maestro      (Huffman     , 0x0ad5c0 - 0x0aff00): removed
 fpf          (Huffman     , 0x0aff00 - 0x0b3c00): removed
 hci          (Huffman     , 0x0b3c00 - 0x0b4980): removed
 fwupdate     (Huffman     , 0x0b4980 - 0x0ba540): removed
 ptt          (Huffman     , 0x0ba540 - 0x0daf40): removed
 touch_fw     (Huffman     , 0x0daf40 - 0x0e6000): removed
The ME minimum size should be 319488 bytes (0x4e000 bytes)
Checking the FTPR RSA signature... VALID
Done! Good luck!
```

privacyguy123, 2020-06-16:
Diving deeper ...

```
Laptop ➜ .\Flash_H.bat

C:\Users\Laptop\Downloads\Razer Drivers\RazerUpdater_v1.1.1.2_D17-2-MEv120381430# cd /d "C:\Users\Laptop\Downloads\Razer Drivers\RazerUpdater_v1.1.1.2_D17-2-MEv120381430\"

C:\Users\Laptop\Downloads\Razer Drivers\RazerUpdater_v1.1.1.2_D17-2-MEv120381430# FWUpdLcl64.exe -f ME_12.0_Consumer_CNP-H_PCH_Production.bin -y -FORCERESET -allowsv
Intel (R) Firmware Update Utility Version: 12.0.38.1430
Copyright (C) 2005 - 2019, Intel Corporation. All rights reserved.

Checking firmware parameters...

Warning: Do not exit the process or power off the machine before the firmware update process ends.

Error 509: Mandatory partitions (FTPR / NFTP / RBEP) were not found in the Update Image.
```

privacyguy123, 2020-06-22:
```
.\FWUpdLcl64.exe -f edited.bin -y -FORCERESET -allowsv
Intel (R) Firmware Update Utility Version: 12.0.38.1430
Copyright (C) 2005 - 2019, Intel Corporation. All rights reserved.

Checking firmware parameters...

Warning: Do not exit the process or power off the machine before the firmware update process ends.

Error 509: Mandatory partitions (FTPR / NFTP / RBEP) were not found in the Update Image.
```

So I whitelist the modules it's complaining about then:

```
.\me_cleaner.py -w FTPR,NFTP,RBEP .\ME_12.0_Consumer_CNP-H_PCH_Production.bin -O edited2.bin
ME/TXE image detected
Found FPT header at 0x10
Found 15 partition(s)
Found FTPR header: FTPR partition spans from 0x2000 to 0xe6000
Found FTPR manifest at 0x2250
ME/TXE firmware version 12.0.38.1430 (generation 3)
WARNING Unknown public key 3efc26920b4bee901b624771c742887b
        Assuming Intel ME
        Please report this warning to the project's maintainer!
Reading partitions list...
 PSVN (0x00001000 - 0x000001100, 0x00000100 total bytes): removed
 UEP  (      no data here      , 0x00000000 total bytes): nothing to remove
 FTPR (0x00002000 - 0x0000e6000, 0x000e4000 total bytes): NOT removed
 FTUP (      no data here      , 0x00000000 total bytes): nothing to remove
 DLMP (      no data here      , 0x00000000 total bytes): nothing to remove
 IVBP (0x000e6000 - 0x0000ea000, 0x00004000 total bytes): removed
 MFS  (0x000ea000 - 0x00014e000, 0x00064000 total bytes): removed
 NFTP (0x0014e000 - 0x000257000, 0x00109000 total bytes): NOT removed
 ROMB (      no data here      , 0x00000000 total bytes): nothing to remove
 UTOK (0x00257000 - 0x000259000, 0x00002000 total bytes): removed
 HVMP (0x00259000 - 0x00025900c, 0x0000000c total bytes): removed
 RBEP (0x0025a000 - 0x000264000, 0x0000a000 total bytes): NOT removed
 RSTR (0x00264000 - 0x000264018, 0x00000018 total bytes): removed
 FLOG (0x00265000 - 0x000266000, 0x00001000 total bytes): removed
 PMCP (0x00266000 - 0x000274000, 0x0000e000 total bytes): removed
Removing partition entries in FPT...
Removing EFFS presence flag...
Correcting checksum (0xd6)...
Reading FTPR modules list...
 FTPR.man     (uncompressed, 0x002250 - 0x003000): NOT removed, partition manif.
 rot.key      (LZMA/uncomp., 0x003000 - 0x003440): removed
 kernel       (Huffman     , 0x003440 - 0x012100): NOT removed, essential
 syslib       (Huffman     , 0x012100 - 0x025180): NOT removed, essential
 bup          (Huffman     , 0x025180 - 0x049540): NOT removed, essential
 pm           (Huffman     , 0x049540 - 0x04c2c0): removed
 vfs          (Huffman     , 0x04c2c0 - 0x05b080): removed
 evtdisp      (Huffman     , 0x05b080 - 0x05db00): removed
 loadmgr      (Huffman     , 0x05db00 - 0x062400): removed
 busdrv       (Huffman     , 0x062400 - 0x064ec0): removed
 gpio         (Huffman     , 0x064ec0 - 0x066bc0): removed
 ipc_drv      (Huffman     , 0x066bc0 - 0x0694c0): removed
 prtc         (Huffman     , 0x0694c0 - 0x06a3c0): removed
 policy       (Huffman     , 0x06a3c0 - 0x06fcc0): removed
 crypto       (Huffman     , 0x06fcc0 - 0x09ca40): removed
 heci         (Huffman     , 0x09ca40 - 0x0a2380): removed
 storage      (Huffman     , 0x0a2380 - 0x0ab2c0): removed
 pmdrv        (Huffman     , 0x0ab2c0 - 0x0ad5c0): removed
 maestro      (Huffman     , 0x0ad5c0 - 0x0aff00): removed
 fpf          (Huffman     , 0x0aff00 - 0x0b3c00): removed
 hci          (Huffman     , 0x0b3c00 - 0x0b4980): removed
 fwupdate     (Huffman     , 0x0b4980 - 0x0ba540): removed
 ptt          (Huffman     , 0x0ba540 - 0x0daf40): removed
 touch_fw     (Huffman     , 0x0daf40 - 0x0e6000): removed
The ME minimum size should be 2527232 bytes (0x269000 bytes)
Checking the FTPR RSA signature... VALID
Done! Good luck!
```

skochinsky, 2020-06-22:
`FWUpdLcl `will not work work with a cleaned image. You need to flash it directly to the chip, ideally with a programmer. FPT *might* work if you disable ME region protection.

privacyguy123, 2020-06-22:
What are the chances I totally brick my machine using said tool?

privacyguy123, 2020-06-22:
`FPT` spititng this out now:

```
.\FPTW64.exe -f fulldump.bin
Intel (R) Flash Programming Tool Version: 12.0.49.1536
Copyright (C) 2005 - 2019, Intel Corporation. All rights reserved.

Reading HSFSTS register... Flash Descriptor: Valid

    --- Flash Devices Found ---
    GD25B128C    ID:0xC84018    Size: 16384KB (131072Kb)

Warning: Unable to detect ME disabled.
Do you wish to proceed with the flash write operation? <Y/N>:
```

Am I good to continue?

skochinsky, 2020-06-22:
> What are the chances I totally brick my machine using said tool?

You are messing with firmware, there is always a chance for bricking the device.

privacyguy123, 2020-06-22:
It seems like the option to disable Intel ME was just hidden in the BIOS - I unhid the option and turned it off ... is there anything else inthere that needs to be set to Disabled?

[image]
[image]
[image]

privacyguy123, 2020-06-23:
Still can't get my head around me_cleaner not playing nice with any of the full dumps I've done with multiple different programs.

```
.\me_cleaner.py -c .\fulldump.bin
Full image detected
The ME/TXE region is valid but the firmware is corrupted or missing.
```

privacyguy123, 2020-06-24:
Now my head is sore - when passing my dumps through `idftool` in Linux it seems happy with setting the HAP bit yet `me_cleaner `just tells me something about a corrupt file. 😕

privacyguy123, 2020-06-27:
My CH341A programmer arrived today and I'm no further on - all the dumps coming off the chip are reported as "corrupt" by `me_cleaner` - this is on ME 12 if that matters.

I unlocked and set the HAP bit with `idftool` and flashed back to the chip successfully and the f**king machine rewrites the chip as part of the bootup process ... WTF?

Stitch626, 2020-08-07:
Yea, some machines do have dual-bios setups where this happens. I've even seen split bios'es across several chips and multiple backups within one chip.

privacyguy123, 2020-11-22:
> 
> 
> Yea, some machines do have dual-bios setups where this happens. I've even seen split bios'es across several chips and multiple backups within one chip.

Any ideas on bypassing this? I have learned this machine does indeed have a backup chip - would I just flash the modded ME to both chips then?

Stitch626, 2020-11-22:
Short answer at the end, but I'd recommend to read everything.

Well, it's kinda hard to provide "remote-support" cuz I was unable to find good pictures of your laptop's motherboard.. 
It would be helpful to get the following information:

How many flashchips has your board? 
That's a bit complicated on high-end-ish boards cuz there are a lot of chips. I'd disassemble the machine until you can take an exact look at the chips (whole board). Then I'd search for SOIC-8 or SOIC-16 chips, labelled with 25xxxx (thats usually SPI-Flash). 
They are usually next to each other if you have more than one, but there are boards in the wild which have for example chips on each side or split between components to keep signal integrity.

Does your board have split-bios or just a dual-bios?
Well, I have an relative old B75 here, which has two 16 MiB flashchips. The bios is split between them and there are backup's of the boot and (kind of) main bios block in the secondary chip (which also contains the ME block). If you have a situation like that, you are most likely screwed cuz there is not much information about it.
There are also "just split" bios setup where your "bios block" is split into two chips which are logically connected as one and so on read as one big chip. That's unusual at consumer hardware from what I've seen so far but it happens.
And ofc there are normal backup setup like some (most?) gigabyte mainboards have..
On these it gets a bit complicated as well. Some check if the main bios chip is just unbootable, some check for broken functions and others verify everything with built in checksums (that's usually on business machines the case).

I hope that the information above is enough and that I didn't miss anything important to explain, let's just continue.

How would I continue if I were you? Well, I'd start with the chips. If you have only one, well, idk, never had any real problems with systems containing only one chip aside of entire dead machines, slow booting machines or hard-off(ish) machines (watchdog timer).
If you have two, I'd read both and kinda "compare" them to each other. They will most likely not exactly match cuz nowadays bios settings are stored as well as the bios itself. They should be let's say about 80% equal to each other (just a guess).
Then you have a dual-bios system with backups (most likely).
If they don't match, you may have a split or "custom" backup bios. Yes, there are backup solutions which are custom, so you don't have a 1:1 copy of the original bios, but fortunately they are relative rare.
If you have a split bios, well, rip you. You could try to put them together (there are even windows cmd commands to do so) and try to open/edit them with the mentioned tools (or with UEFITool, avail on github). There are two variations, so you may be lucky :)

So, let's move on.
One point I should have mentioned earlier but totally forgot about is possible problems with reading/writing of the chip. 
You MAY be able to fully read/write them without desoldering, but it could happen you are not. I personally flash chips on windows (with that chinese software you can get for your CH341A programmer) for a simple reason: less problems.
Lemme explain: If you connect your flasher (however you do it) to the SPI chip, flashrom just does its job.  But thee chinese software reports errors and a good indicator for problems is the "detect flash" function (left side). If it detects your chip, fine. You could get write problems but read is most likely fine. If it doesn't, hm. You are most likely screwed. Unfortunately. 
I hope this doesn't happen to you cuz this either requires you to do dangerous things or just desolder the chip.
Whatever, I did indeed read the whole topic above but you didn't mention anything in special of I missed it.
If you can read from the chip (with detected chip) you should as the next step verify your dump's size. That's a thing most newbies forget about. I've mentioned the chip size above with MiB. Yea, thats correct. MB isn't really specified for x1000 or x1024 measures. But you need to measure exactly with 1024 multiplier. Like 8 MiB means 8192 KibiBytes (KiB). 

This answer got so long and I may have missed some parts but well. To answer your original question in short: Yes. If you have a simple dual bios with two "identical" (80% rule, see above) chip contents, just flash both and see what happens. But be careful, always keep one or more backups of your original contents just in case - even once you are done. Sometimes disabling the ME thing breaks other things.
I tried to give you as much information as possible without leaving too many knowledge-holes (german: wissenslücken) so that you don't accidently brick your machine by not knowing enough.