corna/me_cleaner #331: Unexpected output when using intelmetool

CodeDragon5793, 2020-06-26:
When running `./intelmetool -m` on my computer as root (not sudo either) and with the `iomem=relaxed` boot paramter set in `/etc/default/grub` I receive the following unexpected output:

```Bad news, you have a `Sunrise Point LPC Controller/eSPI Controller` so you have ME hardware on board and you can't control or disable it, continuing...

Can't find ME PCI device```

As far as I can tell, there is no description of what this means. I attempted and succeeded in setting the HAP bit awhile back on my Lenovo ThinkPad T480. That too had some output that was unexpected, but the important part was that it returned that the HAP bit was set. I haven't dumped the firmware and checked since though. When I get a chance I will redump my firmware, check again, and post the output here.

privacyguy123, 2020-06-30:
Does your audio output work with the HAP bit set?

nariox, 2020-07-11:
Mine does (on a T480). Was yours borked? Did you try resetting the BIOS. Somehow mine still shows AMT as enablable even after setting the HAP bit, no idea what would happen if I enabled it.

CodeDragon75, 2021-03-28:
I have since had to return my T480 to Lenovo for a warranty repair. I've had the keyboard backlight fail 3 times on this machine - even without the HAP bit set. Upon returning the machine to me, the repair paper said something about the Firmware. I just checked, and my fears confim that Lenovo reenabled the stupid ME. I will go back through the process again of reflashing the ME. I'd also like to expirement with Coreboot on this as well. I hear that there is a project to get coreboot running on modern, Intel-ME enabled boards.

CodeDragon75, 2021-03-28:
> Does your audio output work with the HAP bit set?

Yes it did.

CodeDragon75, 2021-03-28:
> Mine does (on a T480). Was yours borked? Did you try resetting the BIOS. Somehow mine still shows AMT as enablable even after setting the HAP bit, no idea what would happen if I enabled it.

I borked mine twice when using all options except the `--soft` flag (I can't remember if that is -S or -s right now). I couldn't phyiscally remove Intel ME modules, but setting the HAP bit to 1 worked.