corna/me_cleaner #332: ME not disabled after ( -s ) (update flashed internally with FWUpdLcl)

iGeek098, 2020-07-08:
actually the AltMeDisable bit is set in the image, however, the engine still comes up normally (intelmetool -m shows everything works fine !)

nariox, 2020-07-08:
The BIOS blocks writing to the ME region (on most cases AFAIK). You have to use an external flasher.

iGeek098, 2020-07-09:
No actually ME updated works fine 
(it is a ME updater module that works in ME mode (it has the same permissions as the ME processor so it has access, however it doesn't allow writing a vandalized version of ME (used for updating)

anyway that wasn't the problem, Thanks

iGeek098, 2020-07-09:
the problem is a ME with me_cleaner -S , doesn't update at all, however with me_cleaner -s, it actually changes only a single bit in the FD Descriptor, which doesn't get updated with FWUpdLcl tool

so thanks

closed 2020-07-09 by iGeek098