corna/me_cleaner #336: ME firmware can be fully removed on (some?) Bay Trail boards

kukrimate, 2020-08-01:
I discovered this while porting coreboot to a Bay Trail board. It's a GBYT4-4L by Shenzhen Bostrontium Teng Technology, the board has a Celeron J1800 on-board. I tested the board without a TXE image (ME section completely empty), and I can confirm that it boots just fine and there is no 30 minute watchdog timer. If this can be confirmed on other Bay Trail boards, an option could be added to fully remove the ME on this platform.

Stitch626, 2021-07-11:
Same seems to be true for X79 :)

bol-van, 2022-01-19:
I've had  XCY mini PC with bay trail platform.
System does not boot with stock bios and stripped me.
Only soft disable works. TXEInfo64 cannot communicate with TXE in this state although there's something on the PCI bus

00:1a.0 Encryption controller: Intel Corporation Atom Processor Z36xxx/Z37xxx Series Trusted Execution Engine (rev 0c)
        Subsystem: Intel Corporation Device 7270

prorvazz, 2022-11-06:
Friends, does anyone have a bios for gbyt4-l4 accidentally deleted stock rom. Now the pc turns on but there is no video signal and no boot status.

Ackleys, 2022-12-01:
hi friends,I am a staff member of  Shenzhen Bostrontium Teng Technology ,please contact me if you need support.

prorvazz, 2022-12-01:
@Ackleys Hi! How to contact you? you have hidden contact information in your profile