corna/me_cleaner #340: ME cleaner still working for newer platform

c0d3z3r0, 2020-09-13:
I've been experimenting hours on IFWI image cleaning and finally found out, that IFWI support is not required for me_cleaner to work (#213, #242). The trick is to apply me_cleaner to just the ME subregion of the IFWI image. That can be achieved in this way:

- Get the FIT tool for your ME version
- Open your image with FIT
- Save the project via File>Save
- Apply me_cleaner to <fit tool location>/<image name>/Decom/ME Sub Partition.bin
- Reopen the project (the previously saved xml file, not the original image!)
- Click Build
- Flash the new image and pray ;-)

This was tested with CSME 14.0.11.1205, FIT 14.0.11.1205.

Note: Most modules seem to be mandatory on newer ME versions and the system will reset instantly when one of those gets cleaned (for me at least fpf,maestro,pm,crypto,heci,policy,storage,...). At least I could clean touch_fw,fwupdate,ptt. So, there is not much left to be cleaned and thus, this is not yet a real alternative to #282.

Note 2: you will need the modules whitelist patch from #160

~~~sh
mv "ME Sub Partition.bin" "ME Sub Partition.bin.orig"
me_cleaner.py -O "ME Sub Partition.bin" "ME Sub Partition.bin.orig" -w PSVN,UEP,IVBP,MFS,UTOK,HVMP,RSTR,FLOG,IMDP,RBEP,FTPR,NFTP,DLMP -R touch_fw,ptt,fwupdate
~~~

Yannik, 2020-12-27:
Hi @c0d3z3r0, thanks for the guide. 

Where can I get this "FIT" tool?

steelpuxnastik, 2021-12-27:
> Hi @c0d3z3r0, thanks for the guide.
> 
> Where can I get this "FIT" tool?

FIT - Flash image tool. Find the one that compitable with your processor.

Nephiel, 2022-01-30:
FWIW, this is what worked for me on a Comet Lake with Intel ME 14.x.
**Do this at your own risk**.

- Get `me_cleaner.py` from #282 and ensure it reports your original `dump.bin` as `generation 4` at least. Mine was falling into `gen 3` as the default case so I arbitrarily added a `gen 5` to catch it.
- Use the FIT tool compatible with your processor platform to open your original `dump.bin`
- In the `ME Kernel` section of FIT, find the `Reserved` value (this is the HAP bit, see [reference](https://www.ptsecurity.com/ww-en/analytics/disabling-intel-me-11-via-undocumented-mode/)) and toggle it to `Yes`. Do not change anything else.
- Build the image using FIT but **do NOT flash that**.
- Instead, use `ifdtool -c` on both the original dump and the one built with FIT, and `diff` the outputs. Only one bit should have changed, and this will tell you the offset. Let's say it looks like this:
```
< PCHSTRP28 : 0x801001b8
---
> PCHSTRP28 : 0x801101b8
```
So the HAP bit is the 5th on `PCHSTRP28`.

With this information, we can modify `me_cleaner.py`, editing the `fspba` offset on the `gen 4` (or `5`) branches to replace `PCHSTRP32` with `PCHSTRP28` and `fspba` offset `0x80` with 28x4 = 112 = `0x70`.
In my case, as the fipped bit was still the 5th most significant one, the mask was still `16`, no need to change that.
- Use this `me_cleaner.py` on the original dump, to flip the HAP bit.
- Now, just to check, `hexdump` and `diff` both the original dump and the one just cleaned. **Only that single bit should have changed between them**.
  - (`diff`ing with the one built with FIT should show more changes, this is why we don't want to flash that).
- Flash the dump modified with `me_cleaner.py` and check that ME is disabled sucessfully.

c0d3z3r0, 2023-09-27:
@Nephiel sorry for the late response :D Well, what you describe is actually the soft-disable strap. What I described initially was actual cleaning/wiping of modules to prevent their code to run, even if HAP would had a backdoor. That's what `clean`(er) in `me_cleaner` stands for.