corna/me_cleaner #341: Intel ME still shutting down after CPU upgrade despite being disabled

gameblabla, 2020-09-16:
I own a laptop, the Samsung NP350E7C. It originally came with a Pentium B980 and has an HM70 chipset.

I flashed the ME & BIOS with a CHA341A flasher and it seemed to work fine.
(I stripped both with me_cleaner of course)

I uprade the cpu to an I7-3610QM and this is where the issue happens.

I also upgraded the Intel ME to 8.1.70.1590 and unfortunately, ME still complains about "INVALID CPU AND PCH COMBINATION" and shuts down after 30 minutes, even if the AltMeDisable bit is set with me_cleaner.

I have a several similar issue to this guy
https://github.com/corna/me_cleaner/issues/305

He ""fixed"" it by swapping it with another board that support Ivy Bridge processor but surely there must be another way right ?

Stitch626, 2021-07-10:
Hmm.

From what I've read over the time, the PCH itself contains some configuration data and crypto keys. The manufacturing mode described in your link mentions those FUSE's.
Since the ME only works if you even have a firmware for it, I wonder if it has any watchdog for missing firmware. 
Someone discovered that you have no watchdog on x79 (I know, it's a HEDT platform) when you simply FF the ME firmware area.
I can confirm his findings. Now it would be interesting to know what happens if you FF the ME firmware on thise laptop chipsets (and maybe set the HAP bit).
If you don't have any watchdog for a missing firmware, there should in theory no software left to check for a "licensed" cpu.

Another possibility would be to "hard-disable" the ME-CPU. Some time ago I read about a pin on the PCH which is capable of doing so.

And even another possible path would be to analyze those china mainboards made for x79. Some use even notebook chipsets instead of a proper x79 PCH. They likely had to mod something in regards of the IME.

ppetkov90, 2023-01-06:
I`ve been looking recently for a solution of the problem that @gameblabla had but so far unsuccessfully. Do you guys know if there is a way to use a core i cpu in samsung np300 laptop with HM70 chipset. Has anyone tried the methods mentioned by  @Stitch626  ?

Stitch626, 2023-01-06:
So far: No. I don't have many laptops to test this.
If you (or someone else) is curious about the (dead) link posted in the first post, its available via archive.org:
http://web.archive.org/web/20210623063637/https://blog.ptsecurity.com/2018/10/intel-me-manufacturing-mode-macbook.html

kabal-pl, 2023-12-09:
Hello,
It's been a while, but I'll share my experiences here on this topic.
I have laptop ASUS X401A-WX095V (intel HM70 chipset) - originally came with Celeron B820. 2 weeks ago I swapped CPU to i5-2410M (from Toshiba Satellite Pro C660 on HM65 chipset) and like other Users every 30 min laptop suddenly turned off. But...

Yesterday I found a solution (for me and my x401a) which is as follows:

    1. Dumped BIOS from EEPROM using external programmer (I using latest BIOS for this laptop ver 212), no need to desolder,
    2. Next used Hex editor for downgrade ME Region to 7.0.4.1197. I used stock/clean ME Region 7.0.4.1197 for this downgrade purpose,
    3. I flashed back new modded BIOS with new ME Region,
    ..and after that, laptop was able to turned on, enter to BIOS, but couldn't boot from any connected media drive (ssd, usb)..., so I applied Linux version of "corna / me_cleaner " from github.
    So last step was:
    4. I runed "me_cleaner.py -S" script on my new modded BIOS with downgraded ME Region 7.0.4. In my case works first suggested option from me_cleaner instruction with capital S.

Thats it!!
After that my laptop is able to boot from any media drive and working without 30 min shutdown!

note:
-my Asus X401A on HM70 had original BIOS v212 with ME Region 8.0.10.1464
-donor Toshiba C660 on HM65 had BIOS v1.50 with ME Region 7.0.4.1197

Maybe this will help those who are still looking for a solution. Hope it does ;)

Best regards from Poland!