corna/me_cleaner #342: me_cleaner is outdated

MatteoRaso, 2020-09-26:
I tried using this on my laptop only to be told that the firmware was corrupted every time, despite me verifying that the programmer read the flash rom correctly. Apparently, newer CPUs have slightly different firmware which me_cleaner can't handle. A user named dt-zero made a fork to fix this that worked correctly on my laptop, but his pull request has not been accepted. I know that this probably won't be fixed, since I think that this project has been more or less abandoned, but I thought it was worthwhile to open an issue so that people have trouble will know where to go to get a working me_cleaner fork.

lukejohnsonrp, 2020-09-28:
I think it's now safe to assume that Corna has been threatened by some federal agency, based on his zero activity in this repository since late-2018 (not even a single comment) while being active for other completely non-relevant projects, which you can observe in his Github activity. Reminds me somehow of the Truecrypt story, however I believe their methods are more subtile now and at least they didn't ask him to post some nonsense announcement such as Intel ME is safe and should not be removed so this project is now closed (referring to the event when TrueCrypt author suddenly started recommending BitLocker instead of TrueCrypt, no I did not forget that). We should also not expect all people who get threatened to jeopardize their personal life and go transparent about the threats, so some may just prefer remaining low-profile afterwards as nothing happened.

In fact, NSA and such have not expected having so many people trying to successfully avoid one of their largest secret programs, which could be easily done thanks to this repository. This could also explain how valuable for them would be preventing such a project leader from maintaining that project. Thankfully we are in 2020 and there are still people who won't easily eat their bs and will keep with the project because being FOSS makes it possible. The best we could do now is create the fork and backup important information from the issues such as #3 and similar.

corna, 2020-09-28:
I'm afraid the reason is much simpler: developing me_cleaner requires *a lot* of time, as every test takes many minutes and returns only a binary result (works/doesn't work), and at the moment I have to focus my time on my PhD and related activities. I'd like to resume the work on me_cleaner once I have some spare time, but it won't be soon I'm afraid.

FrostKnight, 2021-02-06:
@corna have you thought of giving other people permission to make releases and do pull requests, since you busy?

It's not a bad idea, I think it would benefit this project immensely.

seniorm0ment, 2021-07-15:
Any updates on this?

FrostKnight, 2021-07-15:
I think it might be a good time to fork the me cleaner... at least until corna finds more time to work on it... which could be a long time or more...  I think its worse, to leave it this way, then to fork it... 

But we would of course need someone with experience to become the new maintainer in the fork. So... yeah. 

Aka, a developer with at least some skills like corna.  But yeah, I support forking this project.  I wish you all the best, anywho... I am sure you guys can figure out what to do regarding these things though.

czeej, 2021-07-17:
@corna 
Best of luck on your PhD. Go kill it.
You contributed enough for now. Let us mortals try to continue with your work :)
Also as something as sensitive as this a bit of secrecy never hurt for development. As old hardware is not changing ;)
Stay safe Corna and godbless.

FrostKnight, 2023-10-16:
> I'm afraid the reason is much simpler: developing me_cleaner requires _a lot_ of time, as every test takes many minutes and returns only a binary result (works/doesn't work), and at the moment I [...]

Do you know when you will be available again for this project? Just curious