corna/me_cleaner #344: "unknown image" for UX32 Asus Zenbook bios

clip123, 2020-12-04:
Thanks for great work on ME Cleaner. I tried this [GitHub guide for Asus H81M-D bios](https://github.com/corna/me_cleaner/issues/62) and me_cleaner.py worked - no issues but I don't have that motherboard and it was for testing purpose only.

Then I grabbed bios for Asus ZenBook series online (listed below) but I get error message for both bios stating "Unknown file" after extracting and running me_cleaner.py script.

Zenbook bois from Asus:
[UX32LNAS204-20190625](https://www.asus.com/us/supportonly/UX32LN/HelpDesk_BIOS/)
[UX32VDAS214-20130307 ](https://www.asus.com/us/Laptops/ASUS-ZenBook-UX32VD/HelpDesk_BIOS/)

mooninite, 2021-03-06:
Same message for the UX301LAA image. The UEFITool is able to read and extract the UEFI image.

Stitch626, 2021-07-11:
I had in the past no luck with UEFITool and ASUS's online bios'es. Those files are incomplete (< 8 MiB) and only meant for updates. Besides that, it's very likely you brick your device while flashing one of them because asus uses some special tools very often and you can't use the bios manufacturer one's.

In simple: use an external flasher, dump the bios-chip, mod the dump, flash the dump.

In long: ASUS's bios update tool only updates partial areas of the flash-chip. They read the DMI configuration out of it, and reflash the bios partially with the DMI data read before. If you now use the bios manufaturer's flash tool, you will loose everything (serial number, lan mac, windows key, etc) and your device will probably be bricked. Unfortunately it's been used of many of their notebooks and the only way to bypass this would be to use an external flasher.