corna/me_cleaner #346: Dell firmware after me_cleaner: Alert! TPM device not detected

freedominside, 2021-01-20:
Hello,

Do you have any idea how to get rid of the message "Alert! TPM device not detected" in the Dell firmware at every boot after using me_cleaner? I have to hit continue everytime I boot the laptop, so the boot process from "power button to login" is interrupted with another key stroke. That would be the only annoyance so far with me_cleaner.

Intel Core i3-6006U
Skylake
Dell Inspiron 15 3567
OEM BIOS
-s flag

TenticleCat, 2021-01-24:
Hello, I would like to apologize for posting an irrelevant subject on your issue. However for many months I have been trying to put me_cleaner on my Dell OEM BIOS. However I have searched from github issues, to phishy sites to not get the information/files needed. Dell Flashing requires an exe instead of a bin file, and every time I attempt to flash a bin file it claims it is invalid. If you are able to share a tutorial on how you ran me_cleaner and flashed a BIOS file, I could be able to help you with your issue, since I also have a Dell Inspiron 15.

Thanks, and apologizes for this irrelevant post.

freedominside, 2021-01-24:
Hello, I followed this tutorial when I used me_cleaner 10 https://github.com/mostav02/Remove_IntelME_FPT

I had to open the laptop and short circuit pin 1 and 5 on the audio chip as described in the tutorial

freedominside, 2021-02-23:
TPM chip is not working when ME is disabled, the alert message cannot be disabled in bios.

closed 2021-02-23 by freedominside

skochinsky, 2021-02-23:
Probably the TPM is implemented in the ME (fTPM: firmware TPM) so you'll have to choose either to live with ME or without TPM (and with the warning).