corna/me_cleaner #347: Thinkpad T410i issues with CPU fan in OS

fearnop-x86, 2021-01-26:
/* Environment */

Code: me_clean Master from 25. Jan. 2021
Python: Version 2.7.16
Flashing HW: Rasperry Pi3 with SOIC8 Pomona Clip
System: Thinkpad T410 (2518A3G)
BIOS: 1.45 (6IET85WW)
CPU: Intel Core i5 M520

/* Tries */

python me_cleaner.py -S -O modified_image.rom t410_1_45.rom: Booting, fan works while booting, but doesnt work in OS
python me_cleaner.py -s -O modified_image.rom t410_1_45.rom: Booting, fan works while booting, but doesnt work in OS
python me_cleaner.py -O modified_image.rom t410_1_45.rom: Booting, fan works while booting, but doesnt work in OS

In all tries the the laptop needs very long time to display something after pressing the power button (~20-40 sec.)
The fan does work while booting the OS, but while staying in the OS it does not. No matter what stress the CPU will have (>100 C)

/* Tested OS */

Linux, systemrescuecd, Version 7.01, Kernel 5.4.78-1-lts
Linux, Tails, Kernel 5.9.0-0.bpo.2

The temp via coretemp-isa-0000 of the cores is correct in both cases and dynamic
The temp via acpitz-acpi-0 seems to be wrong because is stays at +48.0 C, no matter which load the system has.

I've also tested anything also with the original unpatched BIOS and the fan worked all the times while doing exactly the same stress tests. I've reached 100 C CPU core temp without any single round of the fan with the patched BIOS. No matter which options i'm using (-S/-s/no sS)

Stitch626, 2021-06-30:
I had kinda the same result with my X201 some time ago. Unfortunately, my fan didn't even run in the bios. The problem seems to be that the ME firmware kinda manages the thermal control later in the boot process (you may check the other issue about x201, maybe over my replies). I would suggest to either leave the ME as it is (I own a T410, never touched the ME there), replace it with coreboot, use thinkfan or similar software (in my case the acpi access still worked, if I remember correctly), or mod the hardware (pwm controller with temperature sensor).

czeej, 2021-07-17:
Lenovo is not playing as nice as they used to. 
For instance they are even white listing the Wifi card. Meaning if you want to switch to a free wifi card. You have to modify bios anyway.

Stitch626, 2021-07-17:
Yup. But that's not only Lenovo's thing and from what I've seen it all started somewhere in 2005/2006.