corna/me_cleaner #349: Possible new way to disable the me

jakiki6, 2021-02-08:
I've been reading some documents regarding the Intel ME today.

I found something interesting:
_This can be performed by either erasing the descriptor or by corrupting the Flash valid signature at address 0x10_
_Erase the Intel® ME region_
_If CPU/Host has access to Intel® ME region, then one could either erase/program the Intel® ME region to all FFh_

is the 0x10 thing the HAP bit?

ValoWaking, 2021-02-27:
This ALL is not works! The ME and PSP is only visible parts of proprietar processor that include in older chipsets (before haswell) and most of the intels networks. 

You can't disable proprietar processor, and you can't wipe OS of proprietar processor like minix os or threadx os.

Don't wast your life.

me_cleaner is nothing.

Find or create open source hardware if you want be truly freedom!

jakiki6, 2021-02-28:
> Find or create open source hardware if you want be truly freedom!
yeah but it would be nice to disable the ME when you have it already

ValoWaking, 2021-02-28:
no one reason to delete ME. ME is basic security, OC and OC profile part. All bad things is intel AMT. And u can't change anything on platform older then ICH7. Now one of the best way is using ICH7 platform without Intel GBE

jakiki6, 2021-02-28:
Bro they literally describe how to disable the ME in one of their documents about flashing new firmware

ValoWaking, 2021-02-28:
Bro, proprietar processor have his own OS, and he is even works :
1. You PC in standby mode (thanks to ATX PSU standard)
2. You can unsolder you UEFI/BIOS flash chip from your motherboard and this proprietar processor stil work fine even standby mode.

I told u again - ME is only user interface for some OC function

ValoWaking, 2021-02-28:
https://en.wikipedia.org/wiki/Intel_AMT_versions

Stitch626, 2021-07-10:
Well, that's not entirely true. You can disable the processor. And the firmware (minix, on older versions) is stored in the Bios-Flash module. There are machines/boards with separate ME flash-chips, but those are rare. 
There is even a hardware-bit you could set to disable it entirely. But software side is enough. 
Funfact: I own one of those boards with separate ME chips. And do you know what? I have a jumper to disable the entire ME cpu. 
Maybe also read something about Dell's paid "disable" service. They just flash a bios with HAP bit set. HAP environments are (for example) agencies like the NSA or something. They can disable it just like we can.
I'd suggest you better read *more* about the IME cpu in general. It's for example not present on lga 775 chipsets except for the Qxx ones (and some mobile ones as well).