corna/me_cleaner #350: t430 issue.. What's the status of intel me?

i55zvp72xa, 2021-02-09:
Hi. I get this result from intelmetool...

sudo ./intelmetool
Bad news, you have a `QM77 Express Chipset LPC Controller` so you have ME hardware on board and it is very difficult to remove, continuing...
RCBA at 0xfed1c000
MEI not hidden on PCI, checking if visible
MEI found: 7 Series/C216 Chipset Family MEI Controller #1

ME Status   : 0x1e003052
ME Status 2 : 0x1632017a

ME: FW Partition Table      : OK
ME: Bringup Loader Failure  : NO
ME: Firmware Init Complete  : NO
ME: Manufacturing Mode      : YES
ME: Boot Options Present    : NO
ME: Update In Progress      : NO
ME: Current Working State   : Recovery
ME: Current Operation State : M0 with UMA
ME: Current Operation Mode  : Normal
ME: Error Code              : Image Failure
ME: Progress Phase          : BUP Phase
ME: Power Management Event  : Pseudo-global reset
ME: Progress Phase State    : M0 kernel load

PCI READ [bc] : 0x000000bc
ME: Extend SHA-256:***87824433f0a421***

ME has a broken implementation on your board with this BIOS
ME: failed to become ready
WRITE    [00] : CB: 0x80040007
WRITE    [00] : CB: 0x000002ff
ME: failed to become ready
ME: GET FW VERSION message failed
ME: failed to become ready
WRITE    [00] : CB: 0x80080007
WRITE    [00] : CB: 0x00000203
WRITE    [00] : CB: 0x00000000
ME: failed to become ready
ME: GET FWCAPS message failed
exiting

...
And it stops. Is intel me disabled or not? Thank you for this tool.

i55zvp72xa, 2021-02-09:
My bad.. The intel me is gone. Nice!
https://github.com/corna/me_cleaner/wiki/Get-the-status-of-Intel-ME

Keep it up i love tools like these! Thanks!

closed 2021-02-09 by i55zvp72xa

FrostKnight, 2021-11-01:
Yes, Ivy Bridge is currently the best intel processor for freedom without being slow as molasses like sandy bridge  

:)
If I had seen this, I would've told you, yes it is in fact possible to disable intel me in ivy bridge, as I have this already.

Although, a fork might very well be needed, corna seems to be in over his head right now with work... 

I wish he would grant someone else permission to make new releases and pulls. 

But not me, I have no experience with this stuff, just sayin... ;)