corna/me_cleaner #351: How undo me_cleaner?

freedominside, 2021-02-20:
Hello @corna ,

I didn't find this in the wiki. Is there way to undo the changes made by running the me_cleaner -s flag? I.e. going back to stock and reenabling Intel ME? 
Would be on a Dell BIOS with internal flashing method.

Thank you!

corna, 2021-02-20:
>  [data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAEIAAABCCAYAAADjVADoAAAAAXNSR0IArs4c6QAAAARzQklUCAgICHwIZIgAAAAnSURBVHic7cEBDQAAAMKg909tDwcUAAAAAAAAAAAAAAAAAAAAAPBjRFIAASHKmKkAAAAASUVORK5CYII=###24x24:true###][Tracking image][https://github.com/notifications/beacon/ABRJBVEMSFAQ2BE7BAJCJOTS774FPA5CNFSM4X6DUA42YY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4MDQO6SA.gif]

Just re-flash the original firmware

freedominside, 2021-02-20:
Hi, thanks for your answer, but that is not working. EVen if I jump pin 1 and 5 on audio chip again. It flashes the bios but ME stays disabled. I read somewhere that Flash descriptor is usually not flashed during a bios update

freedominside, 2021-02-20:
There is no way to just reflip the hap bit with me_cleaner?

freedominside, 2021-02-21:
My laptop had the Intel Management Engine v11 running, and I disabled it with the HAP bit (-s). Do you think I can adapt your python script to flip the bit again and thus enabling the ME again?

corna, 2021-02-22:
If you do not have the original dump, you can just flip [this bit](https://github.com/corna/me_cleaner/blob/master/me_cleaner.py#L885) (`pchstrp10 &= ~(1 << 7)`) and you should have the original dump back.

freedominside, 2021-02-22:
That sounds good! Unfortunately I don't know how to make the script. What would that script look like to flip it back? I'd be happy to write a little how to for your wiki as a thank you. I think it's always good to include "a way back" in such utilities, obviously only possible when -s was used.

corna, 2021-02-22:
It is generally not possible to roll-back the modifications done by me_cleaner, you have to keep a backup around.
To "make the script" you just have to modify that line with `pchstrp10 &= ~(1 << 7)` and re-run me_cleaner with the -s flag on the modified dump.

freedominside, 2021-02-22:
Thank you. I will try it and report back.

freedominside, 2021-02-22:
I adapted the script and ran it over the .bin, but after running me_cleaner -c I still get `The HAP bit is SET`. Is that normal? I didn't flash it yet.

C:\Users\win10\Desktop\me_cleaner-master>me_cleaner.py -c 213hap.bin
Full image detected
Found FPT header at 0x1010
Found 11 partition(s)
Found FTPR header: FTPR partition spans from 0x1000 to 0xa8000
Found FTPR manifest at 0x1478
ME/TXE firmware version 11.8.65.3590 (generation 3)
Public key match: Intel ME, firmware versions 11.x.x.x
The HAP bit is SET
Checking the FTPR RSA signature... VALID

freedominside, 2021-02-22:
I just modified line 880 with `pchstrp0 &= ~(1 << 16)` as I guess Intem ME v11 is gen3?

This seems to work because me_cleaner -c reports then

C:\Users\win10\Desktop\me_cleaner-master>me_cleaner.py -c 213hap.bin
Full image detected
Found FPT header at 0x1010
Found 11 partition(s)
Found FTPR header: FTPR partition spans from 0x1000 to 0xa8000
Found FTPR manifest at 0x1478
ME/TXE firmware version 11.8.65.3590 (generation 3)
Public key match: Intel ME, firmware versions 11.x.x.x
The HAP bit is NOT SET
Checking the FTPR RSA signature... VALID

freedominside, 2021-02-22:
Can confirm it worked. Thank you.

closed 2021-02-22 by freedominside